bmorcelli/Launcher #1: Documentation for installation of app (bin) files?

spiff72, 2024-02-11:
Hello,

I was wondering if there were instructions/details on how to install the apps that this would allow you to run separately?

I followed the directions on the readme file, and got to the point where it shows the white screen, but it says no apps loaded.  I'm not clear how to load additional apps.

Thanks!

bmorcelli, 2024-02-11:
Hello,

I just made a wiki Page to explai how this launcher works, and will write these instructions on how/where to find the binaries to use with the Launcher.

I just need some time to write it down.

bmorcelli, 2024-02-11:
Hello.. just made a wiki page explaining the process

[Instructions to find binaries] (https://github.com/bmorcelli/M5Stick-Launcher/wiki/Obtaining-binaries-to-launch)

closed 2024-02-11 by bmorcelli

spiff72, 2024-02-11:
Awesome!  This is very helpful, but this leaves me with one question.  How do you copy to the device and where do they go?

Does this require an external SD card accessory?  The M5StickC Plus2 doesn't have an SD card slot so this is part of where I am confused.  I was thinking that the "SD card" in the descriptions was the internal storage/flash, but now I am not so sure.

Thanks!

bmorcelli, 2024-02-29:
Yes, SD Card Hat is something you must build in order tu use this Launcher with M5StickC's devices.

you can use this project to make your own [SDCard Reader](https://www.thingiverse.com/thing:6459069)

Internal storage doesn't have enaugh space to store app files and applications... so we must use SD cards..