bmorcelli/Launcher #4: Source code for `partition-table_8Mb` and `ota_data_initial.bin`?

reggi, 2024-02-24:
Is it possible to build `[partition-table_8Mb.bin](https://github.com/bmorcelli/M5Stick-Launcher/blob/main/support_files/partition-table_8Mb.bin)`  and [ota_data_initial.bin](https://github.com/bmorcelli/M5Stick-Launcher/blob/main/support_files/ota_data_initial.bin) from source as well?

Also using `arduino-cli` is convenient for building

```sh
arduino-cli compile --verbose --fqbn m5stack:esp32:m5stack_cardputer ./Launcher/Launcher.ino --output-dir ./support_files
arduino-cli compile --verbose --fqbn m5stack:esp32:m5stack_cardputer ./StartApp/StartApp.ino  --output-dir ./support_files
cd ./support_files

esptool -p /dev/cu.usbmodem11201 \
        -b 460800 \
        --before default_reset \
        --after hard_reset \
        --chip esp32s3 write_flash \
        --flash_mode dio \
        --flash_freq 80m \
        --flash_size detect 0x0 bootloader_CP.bin 0x8000 partition-table_8Mb.bin 0xe000 ota_data_initial.bin 0x10000 Launcher.ino.bin  0xf0000 StartApp.ino.bin
``

bmorcelli, 2024-02-26:
You can build the partition-table.bin from source, putting the appropriate partition.csv file in the partitions folder of the board and adding it into the arduino-cli command

This is a part of my workflow, where i build the partition-table.bin
`
file="$baseFolder/$latestVersion/tools/partitions/custom_4Mb.csv "
            cp support_files/custom_4Mb.csv ${file}

            arduino-cli compile --fqbn ${{ matrix.board.fqbn }} -e \
                                --build-property build.partitions=custom_4Mb \
                                --build-property upload.maximum_size=1966080  \
                                --build-property compiler.cpp.extra_flags="${extra_flags}" \
                                ./StartApp/StartApp.ino
`

The ota_data_inicial.bin I don't think it is really necessary... I will do some testing here to maybe delete it...

The source code for the Bootloader is something that I am holding for now, because I need to find a way to easily reproduce it to all the devices... I built the binaries one by one adjusting the configs of ESP-IDF...

bmorcelli, 2024-02-26:
Just posted the files about bootloader and the other files.

reggi, 2024-02-26:
Woah thanks for this! Just love to learn how this works and have access to the source!

closed 2024-02-26 by reggi