bmorcelli/Launcher #7: ideas for firmware

tranquocnhat94, 2024-03-16:
Can we connect to a file manager server via wifi to select the firmware to be installed?

bmorcelli, 2024-03-18:
Hi, it is really a good Idea.. but I'm afraid it would use lots of space due to the usage of wifi, webserver and some others libraries..

Actually I'm working on a second app, where we can manage the SD card from a web UI, and upload files to the M5.. and after that load the bin using the launcher..

Depending on the size of the app, maybe it is possible to add into the launcher itself, but would be something exclusive for Cardputer and M5StickC Plus2, due to the 8mb flash memory.

bmorcelli, 2024-05-02:
Done my friend!! thank you for your ideas!!

closed 2024-05-02 by bmorcelli

tranquocnhat94, 2024-05-03:
Your project is absolutely incredible! I'm deeply impressed by your remarkable work and outstanding achievements