bmorcelli/Launcher #16: Cardputer web server crashes with SD card inserted

l0k1, 2024-05-02:
Loving it so far!

With an SD card inserted, I start up the webserver. As soon as I try to connect with PC, the cardputer starts running whatever OTA program is installed and the web server is not available. It doesn't make it as far as the login screen.

With no SD card inserted, it works as expected.

AssetBurned, 2024-05-06:
+1 here. The web UI works fine without my SD card, but if it is insert, the Cardputer just reboots as soon as i try to load the webpage in a browser.

bmorcelli, 2024-05-06:
You are using in "AP mode" or in "my network"?

l0k1, 2024-05-07:
> You are using in "AP mode" or in "my network"?

I'm using "My Network", I haven't managed to get AP mode to work, but I'm probably doing something wrong.

> other thing, are you starting the webUI plugged on USB and the power switch turned off?

I initially tried it running on battery power only, but I did try it with USB and power switch off at your suggestion, and there was no change in the behavior.

> Can you try again, but logging the Serial Monitor, and send me the results?

```
[FROM M5Burner] COM5 opened.
E (50119) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (50119) task_wdt:  - async_tcp (CPU 0/1)
E (50119) task_wdt: Tasks currently running:
E (50119) task_wdt: CPU 0: IDLE
E (50119) task_wdt: CPU 1: async_tcp
E (50119) task_wdt: Aborting.

abort() was called at PC 0x420310a0 on core 0

Backtrace: 0x40377342:0x3fc95bd0 0x4037cda1:0x3fc95bf0 0x40382c6d:0x3fc95c10 0x420310a0:0x3fc95c90 0x403789c1:0x3fc95cb0 0x420ea40f:0x3fcf4b80 0x42031905:0x3fcf4ba0 0x4037e348:0x3fcf4bc0

ELF file SHA256: bd595f1b2c910f84

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376f50
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1024
load:0x403c9700,len:0xb7c
load:0x403cc700,len:0x2dd4
entry 0x403c98a4
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
```

bmorcelli, 2024-05-08:
Few more questions: 
SdCard size:
Formatted in:

Are there files, in your sd card, with some sort of special characters?

l0k1, 2024-05-08:
SD Card size is 32gb, formatted as FAT32. It was a fresh SD card used straight from the packaging into the Cardputer. The only files on it are those placed there by M5Launcher via the OTA download.

```
$ > find .
.
./M5Launcher.txt
./downloads
./downloads/Bruce for Cardputer.bin
./downloads/Car Dashboard for Cardputer.bin
./downloads/CardSkimmerDetector-M5Cardputer.bin
./downloads/CrowOs.bin
./downloads/Evil Portal for Cardputer.bin
./downloads/Interactive SSH client and VPN.bin
./downloads/M5Cardputer_WebRadio.bin
./downloads/NEMO For Cardputer.bin
./downloads/Palnagotchi.bin
./downloads/M5Launcher Cardputer.bin
./downloads/M5 Cardputer User Demo Plus.bin
./downloads/M5Cardputer User Demo Original Firmware 2024.04.23.bin
./System Volume Information
./System Volume Information/WPSettings.dat
./System Volume Information/IndexerVolumeGuid
```

AssetBurned, 2024-05-09:
Likewise here. "My Network" and a 32GB FAT32 card. I do have a mix of filenames with dashes and spaces as well as multiple dots. But nothing that isn't ASCII. I do however have plenty of files starting with a dot.

bmorcelli, 2024-05-12:
@l0k1 @AssetBurned 

Can you please test this version for me? 
[Launcher_V3_Cardputer.zip](https://github.com/bmorcelli/M5Stick-Launcher/files/15284593/Launcher_V3_Cardputer.zip)

The OTA Function is pointing to a test repo, so it won't have all firmwares and wont be updated, but the goal here is undersatand what's going on with your devices.. so I Ask you to use http://web.esphome.io to install this .bin and keep the LOG opened while you try to start the WebUI.

My hunch is that your SDCard is taking too long do deliver all data to the webUI, and the watchdog is stopping the AsyncTCP Service, leading to a crash (at least is what is written in here) 

```
E (50119) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (50119) task_wdt:  - async_tcp (CPU 0/1)
E (50119) task_wdt: Tasks currently running:
E (50119) task_wdt: CPU 0: IDLE
E (50119) task_wdt: CPU 1: async_tcp
E (50119) task_wdt: Aborting.
```

I included some this code soon after starting the server and fixed the fail on creating folders, so if you can test it, would be very nice..
```
  disableCore0WDT(); // disable WDT
  disableCore1WDT(); // disable WDT
  disableLoopWDT();  // disable WDT
```

I Hope it works.. fingers crossed

l0k1, 2024-05-12:
That worked successfully! I installed per your instructions using http://web.esphome.io/. I was able to launch the webserver, upload/download files to the SD card, and install a .bin. Here's the log from that website in case you wanted it.

```
[17:32:28]ESP-ROM:esp32s3-20210327
[17:32:28]Build:Mar 27 2021
[17:32:28]rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
[17:32:28]Saved PC:0x420ea7e2
[17:32:28]SPIWP:0xee
[17:32:28]mode:DIO, clock div:1
[17:32:28]load:0x3fce3810,len:0x1024
[17:32:28]load:0x403c9700,len:0xb7c
[17:32:28]load:0x403cc700,len:0x2dd4
[17:32:28]entry 0x403c98a4
[17:32:28][boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 21
[17:33:05][ 36456][E][esp32-hal-misc.c:143] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT
[17:38:07][338737][E][esp32-hal-misc.c:121] enableCore0WDT(): Failed to add Core 0 IDLE task to WDT
```

bmorcelli, 2024-05-12:
Great!!!

This is great news!!

 Will polish some extra bytes here with the other issues, next week I'll release the fixed version

closed 2024-05-15 by bmorcelli