bmorcelli/Launcher #19: option to repartition or erase the internal storage

AssetBurned, 2024-05-09:
Hi,

Not sure if that is the right wording. so please let me explain the problem.

I have multiple BIN files that i install via this launcher on the Cardputer. But at some point Microhydra won't install anymore and any attempt means I am getting stuck at the progress bar and a reboot will load the Launcher and i can nicely interact, but Microhydra won't load.
So i need to erase the Cardputer with M5burner and then refresh the launcher. Now i can use the Launcher to install Microhydra again and other images.

I can definitely reproduce this when I am switching between the Microhydra 1.0 image i got via M5burner and the CircuitPython 9.1.0-beta.1 from https://circuitpython.org/board/m5stack_cardputer/ .

CU AssetBurned ;-)

bmorcelli, 2024-05-10:
This problem occures because Circuit python overrites the boot.py file in the FAT partition, so when you  install microhydra again, MH won't remake the boot.py and it will not work as expected.. 

This same problem happens when using UIFlow.. I'll look for a way to format the FAT Partition and let it as an option in the Launcher, may be it'll deal with this problem..

bmorcelli, 2024-05-12:
Hello @AssetBurned ..

I think I managed to format the FAT Partition, can you try it for me??
[Launcher_2_fix_Cardputer.zip](https://github.com/bmorcelli/M5Stick-Launcher/files/15284718/Launcher_2_fix_Cardputer.zip)

- Download the file
- Unzip it
- access: https://web.esphome.io 
- Install the bin file you just unzipped

AssetBurned, 2024-05-14:
Sorry, my cardputer and I are currently separated. I expect it back in my hands later this month. But happy to try as soon as I can.

closed 2024-05-15 by bmorcelli

AssetBurned, 2024-05-26:
Follow up: Yes that seems to fix it :-D thank you!!!