bmorcelli/Launcher #20: UiFlow2.0.bin Update error: 5

leorrr, 2024-05-10:
Problems when trying to install **UIFlow2.0.bin**,
**"Update error: 5"**,
any ideas......
Greetings and congratulations for your great work.

bmorcelli, 2024-05-10:
Hi leorr, thanks for using and enjoying Launcher! I really appreciate this..

This issue occurs due to the partition size of UIFlow applications, it is larger then the OTA partition where Launcher install the applications.. I have an Idea on how to overcome that, so I'm putting it on the list of solutions to implement next week.

Thank you very much for reporting!

leorrr, 2024-05-11:
Ok, thank you very much, and again congratulations for your work, I will wait patiently,
Greetings Leo...

bmorcelli, 2024-05-12:
Hello @leorrr 

[Launcher_2_fix_Cardputer.zip](https://github.com/bmorcelli/M5Stick-Launcher/files/15284723/Launcher_2_fix_Cardputer.zip)

Try this test version of the Launcher...

It is pointing to a Test Repo, so the firmwares may be outdated.. it is only for test.. If it is working, I'll publish the release next week

leorrr, 2024-05-12:
It works very well, error 5 no longer appears and it can be installed after I played for a while on the uiFlow 2 page of m5Stack, and very well, I will continue testing and something comes up, I will let you know.
Thank you.

leorrr, 2024-05-12:
Bad news....MicroHidra has stopped working, it can be installed but it does not start, the other bins that I have work fine...

leorrr, 2024-05-13:
Hello, I've been testing the behavior of uiflow2 again and I've realized that despite being usable for making small pieces of code and uploading it through the USB connection, the system doesn't quite work well. The NVS storage (non-volatile storage), where credentials are stored to upload programs to the microcontroller from the cloud, seems to be inaccessible. Also, a menu that usually starts at the beginning to choose the type of connection (web or USB) and list the apps doesn't start either, probably because the NVS is not accessible.
That's a valid perspective. UIFlow is fantastic for teaching programming to beginners, but MicroHydra might be better suited for users who already have some experience and are looking to do something more productive.
I see, maybe it's asking too much of m5Launcher. Anyway, if given the choice, I think I'd prefer compatibility with microHydra over uiflow.

bmorcelli, 2024-05-13:
Actually the problem with UIFlow is that I couldn't install the vfs storage that cames with the UIFlow, because it comes in a different partition..

I learned how to install spiffs partition though, but this nvs is quite of a challange..

I implemented a way to "erase" the fvs filesystem, so the problem when swapping apps won't happen, but the price for this is losing all data you have saved in the nvs partition, so if you  create a file in microhydra, stored in the flash memory, it will be lost when you install other application.

I was able to connect UIFlow into the computer and make a simple program work.. it doesn't show any fancy boot screen or anything, but works..

test this version and tell me if the problem you reported still exists.

[Launcher_v2_fix_Cardputer.zip](https://github.com/bmorcelli/M5Stick-Launcher/files/15296047/Launcher_v2_fix_Cardputer.zip)

leorrr, 2024-05-13:
Hello again! I've tried out this new version... UIFLOW2 works, and Microhydra does too. The only detail I suppose you should be aware of is that when you reinstall UIFLOW, the main.py file, if it was there, I guess it's due to the file deletion system NVS, isn't as neat as UIFLOW's exclusive setup, but the truth is that testing UIFLOW without losing the functionality of other applications like MicroHydra is more than satisfactory.

A detail of what happens when the boot.py file is executed.

[image]

If UIFLOW utilized the SD card in the same way MicroHydra does, I'm sure everything would work perfectly.

Thank you very much, and again congratulations.
Greetings Leo...

closed 2024-05-15 by bmorcelli