bmorcelli/Launcher #21: Cardputer -> Bruce

VooDooFus, 2024-05-10:
For me, when installing Bruce off of a SD card, Launcher borks and goes in to a boot loop. Oddly OTA install seems to work fine. /shrug/

(Please keep in mind, I am dumb.)

:)

VooDooFus, 2024-05-10:
I forgot to mention, the version of Bruce I am using is from m5Burner v3, not github.

bmorcelli, 2024-05-10:
Hey dude,

Have you downloaded this binary from the OTA function? I ask that, because sometimes the file can become corrupted in the download process (ESPs are  not so reliable sometimes..) 

for that, Im sending you the binary, unzip it and copy the .bin into your SD card and try installing. 
[Bruce0.9.zip](https://github.com/bmorcelli/M5Stick-Launcher/files/15280694/Bruce0.9.zip)

closed 2024-05-15 by bmorcelli