bmorcelli/Launcher #29: UIFlow 2 is not compatible with M5Stick-Launcher

Unixware, 2024-06-06:
Installing the UIFlow2 from its .bin file result in black screen, this does not happed when I burn it via M5Burner

bmorcelli, 2024-06-07:
The Python core of UIFlow runs so you can connect to the computer and make some sketches, but the application that is saved in the vfs partition isn't installed yet..

I'm thinking in a way to make it work.. I will keep the issue opened to keep thinking about a solution.

Unixware, 2024-06-07:
Thanks! Nothing important anyway.

bmorcelli, 2024-06-28:
Now it will be compatible with custom partitioning.. 

https://github.com/bmorcelli/M5Stick-Launcher/wiki/Functionalities-explained

Unixware, 2024-06-28:
Thanks ,just installed the new Launcher,  alas the fix does not seem to work for me, still black screen on UIFlow, 
could be incompatible Cardputer or SD card... who knows , nevermind.

bmorcelli, 2024-06-30:
Hey man, yes indeed... I had it working, and idk what I've done that broke it again..

Sorry

I'll keep working on that

Unixware, 2024-06-30:
Its ok I really appreciate your effort, no need to proof that worked, but that video is quite helpful, as in my case I don't see the 'Installing SPIFFS' , instead goes to "Updatining FAT" maybe is some step I ve missed...

bmorcelli, 2024-06-30:
No, you have not.. It was in the testing, and I haven't changed some names yet..

You did fine, its the partition copying that isn't working anymore =(

closed 2024-07-02 by bmorcelli