bmorcelli/Launcher #31: Device restarts itself while running WebUI

Mr-PauI, 2024-06-08:
I'm aware that there is a button to cause a restart, but it seems as I try to connect, randomly and frequently it has begun to just restart itself. I was running 2.0.0, updated to 2.0.1 but it made no difference. The change to my device that I have made that would possibly effect this would be the number of files in the root directory. It is almost impossible for me to use the WebUI at this point, and has occurred mid file upload causing only partial files to be written to the sdcard.

Is there a hard limit on the number of files we can list with the WebUI? Browsing using the SD interface, and the OTA m5burner functions still work as normal. This restart issue however is frustrating my attempts to test iterations of my firmware wirelessly. It generally happens within 30 seconds after connecting to the WebUI. I switched browsers/machines but the issue still persists (chrome/opera).

bmorcelli, 2024-06-08:
Hey buddy.. this is odd.. but I meed some tech info

Device name:

- Connect your device to a serial Monitor (Arduino IDE, is a good enough), make sure it is connected 
- Reproduce the error
- send me the complete output of the Serial Monitor

With the errors output I'll be able to know what might be the problem

Mr-PauI, 2024-06-09:
Device Name: Cardputer (sorry, shoulda known to include that); the first error, failed to remove core 1 idle task occurs immediately upon launching the webui, the reset/Guru Meditiation Error occurs when I try to connect via the web browser - MOST of the time (I'd say around 80%).

ESP-ROM:esp32s3-20210327
[ 20008][E][esp32-hal-misc.c:143] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40057067  PS      : 0x00060f30  A0      : 0x820242f7  A1      : 0x3fcaae30  
A2      : 0x3fcd8ca8  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3fcb1d24  
A6      : 0x00000000  A7      : 0x302e3233  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x40057064:0x3fcaae30 |<-CORRUPTED

ELF file SHA256: ed5620c66c87decc

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x420ea81a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1024
load:0x403c9700,len:0xb7c
load:0x403cc700,len:0x2dd4
entry 0x403c98a4
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12

bmorcelli, 2024-06-10:
Hello again...

```
disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT

```

This first error is supposed to happen indeed, because I'm disabling all watchdogs when starting the WEBUI due to the AsyncTCP handling that was triggering the wdg timer sometimes.. so nothing to worry about that

```
Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.
```

In the other hand, this error is due to the ESP32 attempting to write more than a variable can store.. it is happening because there are too many files, with big names, in the same folder..

A way to deal with this limitation is would be:
- shortening the file names (18 characters is what you can see in the Cardputer screen in most of the time)
- organizing the files in folders, moving the .bin files to a folder, for example... M5 Launcher can handle that without a problem...

It would be great if the Gameboy emulator was storing the .sav files in a separate folder, so it would avoid having lots of files in the root of SDCard, and enabling folder navigation...

I Will think in a way to deal with this limitation providing the file lists to a javascript function, handling the html response in the browser side, but it will take some time

Mr-PauI, 2024-06-10:
There is a similar limitation in the Gameboy Emulator's file listing code as well. Thanks for the explanation of the problem. I was planning on rewriting the file selection logic for the gameboy emulator to dynamically repopulate a handful of local variables to prevent memory limitations from posing a limit on my file explorer. Folders exploration would allow the users to organize their roms a little better, and also solve this issue. Thanks for the suggestion, I'll add some version of a solution to my road map.

Mr-PauI, 2024-06-11:
I added folder navigation to the Gameboy Emulator last night, copied your colour scheme regarding folders; hopefully this can prevent users from overloading the root directory of their SD card. Ended up using a simplified malloc()/freeall() to reuse the existing paging memory so the file listing can be up to 256 entries of 256 bytes per directory at the moment(I could list more but doubt there is a need). Thanks again for the suggestion, it was about time I rewrote that code.

bmorcelli, 2024-06-11:
It was a pleasure to help you and give insights for your wonderful firmware! 

I have plans to solve the problem you raised, but I'm kind of out of time lately.. Bruce is consuming a good portion of my time, I'm stuck with the Core2 project, and I'm trying something new about the io433 project, to integrate both SDCard and CC1101 in the same "HAT" for my StickC...

I hope things will be done and good in a few weeks

closed 2024-06-11 by bmorcelli