bmorcelli/Launcher #33: Bug Report: Launcher Glitch When Exiting WebUI on Cardputer

Axlfc, 2024-06-15:
First of all, let me say what an amazing project the Cardputer and this Launcher is! I've just started tinkering with it, and I'm excited about building from source and installing new things based on the ESP32 platform on it.

However, I found a small but impactful bug in the Launcher that might affect other users trying out different projects. Here’s what I encountered:

Issue Description:
When I flash the M5Launcher and then install projects like 'Marauder for Cardputer', 'Palnagotchi', or 'MicroHydra for Cardputer' (as far as I tested, this might be happening with all other projects you can install on the Cardputer), and then get back into the M5Launcher with an image flashed pressing 'ok' when the letters show, I experience the following issue:

    Upon starting the Launcher's WebUI and then exiting it by pressing 'ok', the Launcher glitches for a few seconds.
    After the glitch, it unexpectedly starts the previously flashed project (e.g., Marauder) instead of staying inside the M5Launcher.

Expected Behavior:
I believe it should stay inside the M5Launcher after exiting the WebUI, effectively performing a 'RESET' to the main screen of the Launcher, ensuring a smooth return to the Launcher interface. This would allow users to either enter another .ino file or manually reset to load the previously flashed image.

This issue can be quite disruptive, especially for users frequently switching between different projects. Resolving it would significantly enhance the robustness of the Launcher and improve the overall user experience.

Thank you for your hard dedication to this project. I'm looking forward to future updates or any improvements in the building-from-source process!

Best regards

bmorcelli, 2024-06-15:
It's really unexpected.. 

I tried here and I see that this is a problem with watchdogs being triggered.. I'll let it deactivated in the next update 

Thank you for reporting!!

bmorcelli, 2024-06-28:
I changed the way  the list of files is sent to the browser, fortunetly now we will be able to heve more files on screen

closed 2024-06-28 by bmorcelli