bmorcelli/Launcher #39: UIFlow 1 not starting on Core2

fhouzim, 2024-07-05:
I'm new to using M5Launcher and encountered an issue whereby installing and starting UIFlow (1.0) on my Core2 would result in a black screen. The device only responds to the reset button or brute power off. UIFlow works normally on my device outside of the launcher. Below I put a dump from the serial monitor from the start of the launcher to UIFlow's fail. There are two tracebacks once UIFlow starts.
```
[FROM M5Burner] COM5 opened.
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1276
ho 0 tail 12 room 4
load:0x40078000,len:13232
load:0x40080400,len:3076
entry 0x400805f0
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 1
[    13][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   469][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   488][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[   497][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[   508][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
[   795][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[   804][D][M5GFX.cpp:796] autodetect(): [M5GFX] AXP192 found
[   813][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x40108988 arg=0x3ffbdc30
[   823][D][M5GFX.cpp:494] _read_panel_id(): [M5GFX] [Autodetect] read cmd:04 = 7f0000e3
[   832][I][M5GFX.cpp:934] autodetect(): [M5GFX] [Autodetect] M5StackCore2
[   960][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[   970][V][Touch_FT5x06.cpp:63] _check_init(): [FT5x06] CIPHER:0x64 / FIRMID:0x01 / VENDID:0x01
[   979][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[  1039][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(-1) txPin(-1)
[  1048][V][esp32-hal-uart.c:343] uartBegin(): UART0 Driver already installed.
[  1058][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1276
ho 0 tail 12 room 4
load:0x40078000,len:13232
load:0x40080400,len:3076
entry 0x400805f0
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12

       _  __ _               
 _   _(_)/ _| | _____      __
| | | | | |_| |/ _ \ \ /\ / /
| |_| | |  _| | (_) \ V  V / 
 \__,_|_|_| |_|\___/ \_/\_/  

Traceback (most recent call last):
  File "_boot.py", line 56, in <module>
  File "m5stack.py", line 4, in <module>
  File "hardware/__init__.py", line 11, in <module>
  File "hardware/axp192.py", line 5, in <module>
  File "deviceCfg.py", line 7, in <module>
IndexError: list index out of range
Traceback (most recent call last):
  File "_flow.py", line 5, in <module>
ImportError: can't import name power
MicroPython bd726ecdd-dirty on 2024-06-20; M5Stack with ESP32
Type "help()" for more information.
>>> 
```

bmorcelli, 2024-07-10:
I spent a lot of time here trying to figure out a way to make UiFlow1 work with M5Launcher, and here is what I've lerned:
- for UiFlow1 to start it needs to setup a "phy_init" partition, which I was able to reproduce and make it happen
- But it freezes a few seconds in the menu Screen... when it tries to start the WiFi interface.

I was able to star it via USB, but sketches that start WIFI make the ESP32 to freeze, and it only works when I inject the original bootloader, meaning that I would need to make a "custom" bootloader (again) for this case.. not a real problem, but then I found out that:
- UiFlow uses MicroPython 1.12
- Micropithon 1.12 uses ESP-IDF 3.3.x (we are now on 5.3)

And this lat thing leads to a (most probably) huge problem, for two reasons... 
- first: I'm afraid to "rollback" the bootloader and break the WiFi communications to the whole other firmwares
- second: It can be a partition settings that I don't have access (needed to look at their code) 

I'll keep looking onto it.. but probably it'll take a very long time for us to see UiFlow1 working with Launcher :(

fhouzim, 2024-07-11:
I wouldn't mind as much and just use UiFlow 2 but it misses a lot of functionality and stuff that its predecessor has.. it's a shame that UiFlow 1 is such a nuisance to work with in this case.

closed 2025-05-10 by bmorcelli