bmorcelli/Launcher #49: CYD 2USB BL not bright

CoolmdXi, 2024-07-21:
This is an amazing project, saw it on Brian's discord. Installed it on a CYD 2USB and its works very well, with the exception of my screen being very dim, BL doesn't look like its on very high. Even when I checked it in settings it says screen is at 100 %.
Apart from that , do you know of many BIN files for the CYD ?. Have tried the few from brian's discord.
There seems to be lots for the M5 stack

bmorcelli, 2024-07-21:
Indeed the 100% of brightness isn't quite well..

I'll try some other pwm frequencies to control the brightness and make it better.. 

When it's ready, I'll publish in that group agai

CoolmdXi, 2024-07-21:
#define BACKLIGHT_PIN 21  /
 pinMode(BACKLIGHT_PIN, OUTPUT);  //define backlight as output
  ledcWrite(BACKLIGHT_PIN, 255);   // set backlight to max
  // Configure PWM
  ledcAttachChannel(BACKLIGHT_PIN, 5000, 8, 0);
  ledcWrite(BACKLIGHT_PIN, 255);  // set backlight to max

This PWM setup worked well with CYD 2USB on a recent clock I was working on.. maybe it can help

CoolmdXi, 2024-07-21:
I have managed to put one or two of my own sketches onto the Project, as I am fairly new to this.
I used the export bin from Arduino IDE, but confused me with different Bin files from sketches, but worked.
Many thanks for your code I really think it is Good and very useful

bmorcelli, 2024-07-21:
The problem with the brightness was the bit mode I was using..

I just changed it to 8 and made some tweaks on the values for settings to better balance the bright steps..

its now available

closed 2024-07-21 by bmorcelli

CoolmdXi, 2024-07-21:
Ok that's great just flashing it now. Many thanks