bmorcelli/Launcher #51: Port for the FabGL VGA32?

FreeLANMan, 2024-07-22:
Any chance of seeing a port for the FabGL VGA32 card?

bmorcelli, 2024-07-22:
Do you have a link to this board?

I want to port it to T-Deck and T-Embed someday.. but I need the boards to test..

FreeLANMan, 2024-07-22:
The official site:
https://www.lilygo.cc/products/fabgl-vga32

bmorcelli, 2024-11-01:
Hey Brother, I've made a version of Launcher that might run in this device..

 It is the ESP32 4mb headless..

I couldn't figure out if it runs an ESP32 or not, but if it does it will work like this:
- after flashing complete, restart the device (only the first time)
- when power on (or press RST btn) it will wait for 5 seconds you press GPIO 0 (boot btn) btn (idk if this board have this, if not, you can change it here: https://github.com/bmorcelli/M5Stick-Launcher/blob/main/platformio.ini#L962

- it will start a Web User Interface, where you can install your firmware using the OTA Update option, setup the SDCard pins and setup the Wifi network you want it to try to connect

closed 2024-11-25 by bmorcelli