bmorcelli/Launcher #53: Minigotchi porting to M5Launcher

dj1ch, 2024-07-23:
Hello, 

I am the lead maintainer of the Minigotchi project, and I would like to know if I can port this to the M5Launcher. However, there is a major problem that I am unsure of. The Minigotchi has this file called `config.cpp`, essentially a configuration file that controls many parts of the Minigotchi, whether settings like the screen being used or enabling/disabling core features like deauthing. The problem is that it must be modified before flashing the Minigotchi. Of course, it will run fine with default configurations, but things such as the screen may not work. 

The most doable solution in this scenario is probably to make multiple pre-built binaries with default configurations + screen support. I'm not sure if you would have anything else added to allow it to be ported. Most of what needs to be changed anyway is just the configuration for the screen. 

Thanks, dj1ch

bmorcelli, 2024-07-23:
what's up dj1ch.. I've being following your project and i've contributing with it through ATOMNFT by helping him port it to the m5 devices..

To work with the Launcher there's nothing else to be done, other than have the bin file to install.. so sharing a pre-compiled binary is the way

I was talking to ATOMNFT about a way to enhance Minigotchi's project, and he came with the WiFi whitelist problem, and maybe some other problems that must me set in the building process..  My suggestion is allow the user to configure it by: 
- Serial Command interface  and save the configuration on SPIFFS file system, 
- Using a Web Interface to configure this file in the SPIFFS (maybe start it with the Serial Command Interface), which is my favorite solution

Count with me to help in the minigotchi's project

dj1ch, 2024-07-25:
> what's up dj1ch.. I've being following your project and i've contributing with it through ATOMNFT by helping him port it to the m5 devices..
>
> To work with the Launcher there's nothing else to be [...]

In my opinion, I would want to go with the second option, I could probably have some boolean to check whether or not it's the first boot

```cpp
bool firstBoot = true;

Minigotchi::init() {
  if (firstBoot) {
    startWebUi();
    firstBoot = false;
  }
  // rest of startup processes
}
```

But that's just me thinking out loud. I'll implement this in a later release.

dj1ch, 2024-07-26:
> Thinking about ESP32 boards in general, maybe its a good idea to use the "boot" btn to trigger this WebUi, so you could start capturing some EAPOL/PMKID, saving on Spiffs memory and retrieve them with the WebUI..
> 
> Boot btn 99% of the cases is the GPIO 0 (zero)...
> 
> The check can be done at any point of the loop, after minigotchi starts and webUi be launched (and closed using the same logic) at any time...
> 
> With these PMKID and stuff the project is closer to the official pwnagotchi in functionality...

Do all ESP32s have a button that can be used? Or is this just exclusive to specific ESP32s? Sniffing for EOPOL/PMKID shouldn't be a problem if that is the case.

bmorcelli, 2024-07-26:
Almost all ESP devboards have at least one button.. and if it have a tft, it will probably have one too..

Sniffing for EAPOL/PMKID is something easy to find examples over the GitHub..

dj1ch, 2024-08-21:
I'm going to close this as #64 has been merged. Thank you for the help @bmorcelli!

closed 2024-08-21 by dj1ch

bmorcelli, 2024-08-21:
Congratulations brother, for your amazing project!