bmorcelli/Launcher #58: UIFlow 2.0 beeps and reboots in a loop after installing

psifertex, 2024-08-03:
Just updated to the latest launcher and tried the UIFlow2.0 download on my M5Cardputer. It gets stuck in a boot-loop beeping and showing half a screen.

Here's the logs during installation before it goes into that:
```
[ 49296][I][sd_functions.cpp:538] performUpdate(): Update successfully completed.
[ 49346][I][sd_functions.cpp:712] performFATUpdate(): Start updating: vfs
[ 49353][I][sd_functions.cpp:714] performFATUpdate(): Erasing updating: vfs from: 6750208 with size: 524288
[ 49862][I][sd_functions.cpp:724] performFATUpdate(): Updating updating: vfs
[ 52320][I][sd_functions.cpp:740] performFATUpdate(): Success updating vfs
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x420f4356
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x518
load:0x403c9700,len:0xc08
load:0x403cc700,len:0x2a0c
entry 0x403c98dc
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
Terminal disconnected: NetworkError: The device has been lost.
```

I saw this version is only a few days old, but I haven't tried it before so no idea if this is a new issue or not.

bmorcelli, 2024-08-06:
You need to change the partition scheme to "UIFLOW2" 

CFG -> Part Change -> UiFlow2

then it is supposed to work

psifertex, 2024-08-06:
Thanks! Worked for me.

closed 2024-08-06 by psifertex