bmorcelli/Launcher #59: Having issue trying to access sd card via webui

Bulkoa, 2024-08-06:
I am having issue with access the files on sd card via webui my sd card is only 185MB out of 32GB and
the webui still load in my browser but when the file list trying to load, my cardputer just boot the firmware that i installed.

bmorcelli, 2024-08-06:
Do you have too many files in the root folder? (Ive seen it restart when there's more than 40~60 files in the folder, depending on the name size)

Can you repeat the process and send me the Serial Monitor log?

You can use the m5burner serial monitor to do it..

Bulkoa, 2024-08-07:
i think that is a problem with my cardputer i will delete some files that i didn't want.

serial log:

[FROM M5Burner] COM3 opened.
[   314][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200[  4763][I][main.cpp:95] get_partition_sizes(): label: vfs
[  4768][I][main.cpp:106] get_partition_sizes(): [Partition Sizes] MAX_APP: 5111808
[  4775][I][main.cpp:107] get_partition_sizes(): [Partition Sizes] MAX_SPIFFS: 1048576
[  4783][I][main.cpp:108] get_partition_sizes(): [Partition Sizes] MAX_FAT_sys: 0
[  4790][I][main.cpp:109] get_partition_sizes(): [Partition Sizes] MAX_FAT_vfs: 524288
[  4826][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
[ 14630][I][display.cpp:540] loopOptions(): Number of options: 2
[ 17385][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 17440][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Started
[ 17446][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 2 - STA_START
[ 21048][V][WiFiGeneric.cpp:384] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 10
[ 21057][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 21265][I][display.cpp:540] loopOptions(): Number of options: 10
[ 24910][I][settings.cpp:322] getConfigs(): getConfigs: config.conf exists
[ 24928][I][settings.cpp:330] getConfigs(): getConfigs: deserialized correctly
[ 24938][I][settings.cpp:350] getConfigs(): Brightness: 100
[ 24943][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
[ 24954][I][settings.cpp:380] getConfigs(): Using config.conf setup file
[ 24972][I][settings.cpp:322] getConfigs(): getConfigs: config.conf exists
[ 24990][I][settings.cpp:330] getConfigs(): getConfigs: deserialized correctly
[ 25000][I][settings.cpp:350] getConfigs(): Brightness: 100
[ 25005][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
[ 25015][I][settings.cpp:380] getConfigs(): Using config.conf setup file
[ 25023][I][onlineLauncher.cpp:48] wifiConnect(): sdcardMounted: 1
[ 25029][I][onlineLauncher.cpp:54] wifiConnect(): SSID: myNetSSID, Pass: myNetPassword
[ 25037][I][onlineLauncher.cpp:54] wifiConnect(): SSID: Rdfrta_2.4G, Pass: 089089089
[ 25044][I][onlineLauncher.cpp:58] wifiConnect(): Found SSID: Rdfrta_2.4G
[ 25140][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 25638][V][WiFiGeneric.cpp:356] _arduino_event_cb(): STA Connected: SSID: Rdfrta_2.4G, BSSID: c8:5a:9f:bd:32:9c, Channel: 9, Auth: WPA_WPA2_PSK
[ 25651][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[ 28030][V][WiFiGeneric.cpp:370] _arduino_event_cb(): STA Got New IP:192.168.1.11
[ 28037][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[ 28045][D][WiFiGeneric.cpp:1103] _eventCallback(): STA IP: 192.168.1.11, MASK: 255.255.255.0, GW: 192.168.1.1
[ 28914][E][esp32-hal-misc.c:143] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT
E (60193) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (60193) task_wdt:  - async_tcp (CPU 0/1)
E (60193) task_wdt: Tasks currently running:
E (60193) task_wdt: CPU 0: IDLE0
E (60193) task_wdt: CPU 1: async_tcp
E (60193) task_wdt: Aborting.

abort() was called at PC 0x420392a0 on core 0

Backtrace: 0x403772ca:0x3fc94e00 0x4037c1c5:0x3fc94e20 0x403820e5:0x3fc94e40 0x420392a0:0x3fc94ec0 0x4037897d:0x3fc94ee0 0x420f436f:0x3fcf4c20 0x42039b05:0x3fcf4c40 0x4037d76c:0x3fcf4c60

ELF file SHA256: fd86889cd144089d

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376ed8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x518
load:0x403c9700,len:0xc08
load:0x403cc700,len:0x2a0c
entry 0x403c98dc
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12

closed 2024-08-09 by Bulkoa