bmorcelli/Launcher #61: Error sending a file to the SD card through the webUI from iPhone

auroJimen, 2024-08-15:
Hi (first of all thanks for everything m5launcher is hands down the most usefull firmware for the cardputer i've used)

Since the last update (2.2.3) I've noticed that whenever I try to send a file through the webUI from my iPhone the process crasses and M5launcher opens the last firmware you had loaded. I've checked several combinations of settings and files and the issue persists. Sending the file from a PC works normally.

Combinations checked:

- Both AP mode and External wifi
- File extention doesn't seem to matter
- I've tried it w/ the lastest version as per github (2.2.3) and as per M5Burner (2.2.4)

I connected a serial terminal to get the debugging info during the bug:

> [FROM M5Burner] COM4 opened.
> [   308][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200[  9829][I][display.cpp:545] loopOptions(): Number of options: 2
> [ 12214][I][settings.cpp:327] getConfigs(): getConfigs: config.conf exists
> [ 12240][I][settings.cpp:335] getConfigs(): getConfigs: deserialized correctly
> [ 12249][I][settings.cpp:355] getConfigs(): Brightness: 100
> [ 12255][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [ 12265][I][settings.cpp:385] getConfigs(): Using config.conf setup file
> [ 12281][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
> [ 12329][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Started
> [ 12329][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 172.0.0.1, MASK: 255.255.255.0, GW: 172.0.0.1
> [ 12347][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 10 - AP_START
> [ 12355][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 172.0.0.1 | Gateway: 172.0.0.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
> [ 12368][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 172.0.0.2 to 172.0.0.12
> [ 13079][E][esp32-hal-misc.c:143] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT
> [ 25406][V][WiFiGeneric.cpp:408] _arduino_event_cb(): AP Station Connected: MAC: 4e:ca:40:7b:38:52, AID: 1
> [ 25416][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
> [ 25617][V][WiFiGeneric.cpp:422] _arduino_event_cb(): AP Station IP Assigned:172.0.0.2
> [ 25625][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
> [ 27198][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 169
> E (63678) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E (63678) task_wdt:  - async_tcp (CPU 0/1)
> E (63678) task_wdt: Tasks currently running:
> E (63678) task_wdt: CPU 0: IDLE0
> E (63678) task_wdt: CPU 1: loopTask
> E (63678) task_wdt: Aborting.
> 
> abort() was called at PC 0x42038fc8 on core 0
> 
> 
> Backtrace: 0x403772ca:0x3fc94e00 0x4037c1c5:0x3fc94e20 0x403820e5:0x3fc94e40 0x42038fc8:0x3fc94ec0 0x4037897d:0x3fc94ee0 0x420f4097:0x3fcf4c20 0x4203982d:0x3fcf4c40 0x4037d76c:0x3fcf4c60
> 
> 
> 
> 
> ELF file SHA256: 43aa9338518a1735
> 
> Rebooting...
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
> Saved PC:0x40376ed8
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3808,len:0x518
> load:0x403c9700,len:0xc08
> load:0x403cc700,len:0x2a0c
> entry 0x403c98dc
> [boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
> [   489][D][esp32-hal-tinyusb.c:680] tinyusb_enable_interface(): Interface HID enabled
> [   496][D][USBHID.cpp:61] tinyusb_enable_hid_device(): Device[0] len: 67
> E (525) esp_ota_ops: not found otadata
> [   541][D][main.cpp:230] setup(): Total heap: 253840
> [   546][D][main.cpp:231] setup(): Free heap: 221056
> [   551][D][main.cpp:234] setup(): PSRAM Not Found
> [   555][D][main.cpp:235] setup(): Total PSRAM: 0
> [   560][D][main.cpp:236] setup(): Free PSRAM: 0
> [   565][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not have default pins on ESP32S3!
> [  1807][I][main.cpp:177] load_eeprom():   
> *-*EEPROM Settings*-*   
> - rotation  =001,   
> - dimmerSet =015,   
> - Brightness=100,   
> - IR Tx Pin =044,   
> - IR Rx Pin =001,   
> - RF Tx Pin =002,   
> - RF Rx Pin =001,   
> - Time Zone =000,   
> - FGColor   =0xA80F   
> *-*-*-*-*-*-*-*-*-*-*
> [  1833][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [  8873][D][Tone.cpp:124] tone(): _pin=2, frequency=5000 Hz, duration=50 ms
> [  8880][V][Tone.cpp:62] tone_init(): Creating tone queue
> [  8885][V][Tone.cpp:68] tone_init(): Tone queue created
> [  8890][V][Tone.cpp:72] tone_init(): Creating tone task
> [  8895][V][Tone.cpp:85] tone_init(): Tone task created
> [  8900][D][Tone.cpp:31] tone_task(): Task received from queue TONE_START: _pin=2, frequency=5000 Hz, duration=50 ms
> [  8910][D][Tone.cpp:33] tone_task(): Setup LED controll on channel 0
> [  9100][D][Tone.cpp:124] tone(): _pin=2, frequency=5000 Hz, duration=50 ms
> [  9107][D][Tone.cpp:31] tone_task(): Task received from queue TONE_START: _pin=2, frequency=5000 Hz, duration=50 ms
> [  9117][D][Tone.cpp:33] tone_task(): Setup LED controll on channel 0
> [  9338][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x420984d4 arg=0x3fc9c4e4
> [  9363][I][settings.cpp:596] getConfigs(): getConfigs: config.conf exists
> [  9389][I][settings.cpp:608] getConfigs(): getConfigs: deserialized correctly
> [  9396][I][settings.cpp:626] getConfigs(): Brightness: 100
> [  9401][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [  9412][I][settings.cpp:647] getConfigs(): Wrote new conf to EEPROM
> [  9418][I][settings.cpp:649] getConfigs(): Using config.conf setup file
> Enf o Config
> [FROM M5Burner] Serial Port `Disconnected.
>

bmorcelli, 2024-08-15:
What's up friend!! Thank you for this rich report!!

The problem is with the watchdog that get triggered by async_tcp task

```
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (63678) task_wdt: - async_tcp (CPU 0/1)
```

I'm struggling to find a way to reset this timer and make it work properly..

I'm reluctant to change the Webserver library, but I fear this will be the only solution

auroJimen, 2024-08-15:
Thanks for answering so fast, i don't know if there's any way to be of help, but if so let me know!

I'm now thinking that the issue may be an update to safary (?) or some other thing on the other side of the problem. Tomorrow I'll try and check with an older version of m5launcher to see if i can get that feature to work again.

bmorcelli, 2024-08-16:
Can you please test with this version?

[Launcher_2.0_Cardputer.zip](https://github.com/user-attachments/files/16631989/Launcher_2.0_Cardputer.zip)

Install it using [This flasher](https://web.esphome.io/)

I've found a way to remove the Watchdog from AsyncTCP just now..

auroJimen, 2024-08-16:
I've just done it, seems like its throwing the same unfed watchdog error

> [   281][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200[  5443][I][main.cpp:91] get_partition_sizes(): label: vfs
> [  5448][I][main.cpp:102] get_partition_sizes(): [Partition Sizes] MAX_APP: 5111808
> [  5456][I][main.cpp:103] get_partition_sizes(): [Partition Sizes] MAX_SPIFFS: 1048576
> [  5464][I][main.cpp:104] get_partition_sizes(): [Partition Sizes] MAX_FAT_sys: 0
> [  5471][I][main.cpp:105] get_partition_sizes(): [Partition Sizes] MAX_FAT_vfs: 524288
> [  5506][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [ 15530][I][display.cpp:545] loopOptions(): Number of options: 2
> [ 18249][I][settings.cpp:327] getConfigs(): getConfigs: config.conf exists
> [ 18275][I][settings.cpp:335] getConfigs(): getConfigs: deserialized correctly
> [ 18284][I][settings.cpp:355] getConfigs(): Brightness: 100
> [ 18290][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [ 18300][I][settings.cpp:385] getConfigs(): Using config.conf setup file
> [ 18314][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
> [ 18361][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Started
> [ 18361][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 172.0.0.1, MASK: 255.255.255.0, GW: 172.0.0.1
> [ 18378][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 10 - AP_START
> [ 18386][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 172.0.0.1 | Gateway: 172.0.0.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
> [ 18399][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 172.0.0.2 to 172.0.0.12
> [ 18418][V][WiFiGeneric.cpp:396] _arduino_event_cb(): AP Stopped
> [ 18426][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 11 - AP_STOP
> [ 18426][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Started
> [ 18439][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 10 - AP_START
> [ 19123][E][esp32-hal-misc.c:143] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT
> [ 33234][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 169
> [ 42309][V][WiFiGeneric.cpp:408] _arduino_event_cb(): AP Station Connected: MAC: 4e:ca:40:7b:38:52, AID: 1
> [ 42318][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
> [ 43673][V][WiFiGeneric.cpp:422] _arduino_event_cb(): AP Station IP Assigned:172.0.0.2
> [ 43681][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
> E (100500) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E (100500) task_wdt:  - async_tcp (CPU 0/1)
> E (100500) task_wdt: Tasks currently running:
> E (100500) task_wdt: CPU 0: IDLE0
> E (100500) task_wdt: CPU 1: loopTask
> E (100500) task_wdt: Aborting.
> 
> abort() was called at PC 0x420391bc on core 0
> 
> 
> Backtrace: 0x403772ca:0x3fc94e00 0x4037c1c5:0x3fc94e20 0x403820e5:0x3fc94e40 0x420391bc:0x3fc94ec0 0x4037897d:0x3fc94ee0 0x420f4293:0x3fcf4c20 0x42039a21:0x3fcf4c40 0x4037d76c:0x3fcf4c60
> 
> 
> 
> 
> ELF file SHA256: c369db4cb7dc6b22
> 
> Rebooting...
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
> Saved PC:0x40376ed8
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3808,len:0x518
> load:0x403c9700,len:0xc08
> load:0x403cc700,len:0x2a0c
> entry 0x403c98dc
> [boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
> [   310][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(44) txPin(43)
> [   319][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
> [   327][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
> [   336][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not have default pins on ESP32S3!
> [  1576][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [  1705][I][partitioner.cpp:396] partitionCrawler(): [Partitioneer] Running partition is ESP_PARTITION_SUBTYPE_APP_TEST, no action taken
> [  1717][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x4202de90 arg=0x3fc93d20
> [  1752][I][settings.cpp:327] getConfigs(): getConfigs: config.conf exists
> [  1778][I][settings.cpp:335] getConfigs(): getConfigs: deserialized correctly
> [  1788][I][settings.cpp:355] getConfigs(): Brightness: 100
> [  1793][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [  1803][I][settings.cpp:385] getConfigs(): Using config.conf setup file
> [  6862][I][main.cpp:91] get_partition_sizes(): label: vfs
> [  6867][I][main.cpp:102] get_partition_sizes(): [Partition Sizes] MAX_APP: 5111808
> [  6874][I][main.cpp:103] get_partition_sizes(): [Partition Sizes] MAX_SPIFFS: 1048576
> [  6882][I][main.cpp:104] get_partition_sizes(): [Partition Sizes] MAX_FAT_sys: 0
> [  6889][I][main.cpp:105] get_partition_sizes(): [Partition Sizes] MAX_FAT_vfs: 524288
> [  6925][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [ 15271][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 169
> [ 15279][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz

bmorcelli, 2024-08-16:
Ok...

today's last try  [Launcher_2.0_Cardputer2.zip](https://github.com/user-attachments/files/16632213/Launcher_2.0_Cardputer2.zip)
Added `esp_task_wdt_reset();` into the `static void _async_service_task(void *pvParameters)` function on AsyncTCP Library to reset the AsyncTCP periodically

auroJimen, 2024-08-16:
Thanks for giving it one last try today, sadly I got the same result :(

> [  6848][I][main.cpp:91] get_partition_sizes(): label: vfs
> [  6854][I][main.cpp:102] get_partition_sizes(): [Partition Sizes] MAX_APP: 5111808   
> [  6861][I][main.cpp:103] get_partition_sizes(): [Partition Sizes] MAX_SPIFFS: 1048576
> [  6869][I][main.cpp:104] get_partition_sizes(): [Partition Sizes] MAX_FAT_sys: 0     
> [  6876][I][main.cpp:105] get_partition_sizes(): [Partition Sizes] MAX_FAT_vfs: 524288
> [  6912][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [ 14802][I][display.cpp:545] loopOptions(): Number of options: 2
> [ 17769][I][settings.cpp:327] getConfigs(): getConfigs: config.conf exists
> [ 17795][I][settings.cpp:335] getConfigs(): getConfigs: deserialized correctly
> [ 17805][I][settings.cpp:355] getConfigs(): Brightness: 100
> [ 17810][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [ 17821][I][settings.cpp:385] getConfigs(): Using config.conf setup file
> [ 17836][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
> [ 17877][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Started
> [ 17877][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 172.0.0.1, MASK: 255.255.255.0, GW: 172.0.0.1
> [ 17895][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 10 - AP_START
> [ 17903][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 172.0.0.1 | Gateway: 172.0.0.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
> [ 17915][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 172.0.0.2 to 172.0.0.12
> [ 18626][E][esp32-hal-misc.c:143] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT
> [ 26965][V][WiFiGeneric.cpp:408] _arduino_event_cb(): AP Station Connected: MAC: 4e:ca:40:7b:38:52, AID: 1
> [ 26974][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
> [ 27261][V][WiFiGeneric.cpp:422] _arduino_event_cb(): AP Station IP Assigned:172.0.0.2  
> [ 27268][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
> [ 32753][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 169
> E (61308) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E (61308) task_wdt:  - async_tcp (CPU 0/1)
> E (61308) task_wdt: Tasks currently running:
> E (61308) task_wdt: CPU 0: IDLE0
> E (61308) task_wdt: CPU 1: loopTask
> E (61308) task_wdt: Aborting.
> 
> abort() was called at PC 0x42039210 on core 0
> 
> 
> Backtrace: 0x403772ca:0x3fc94e00 0x4037c1c5:0x3fc94e20 0x403820e5:0x3fc94e40 0x42039210:0x3fc94ec0 0x4037897d:0x3fc94ee0 0x420f42e7:0x3fcf4c20 0x42039a75:0x3fcf4c40 0x4037d76c:0x3fcf4c60
> 
> 
> 
> 
> ELF file SHA256: c2f59ef09a4dd110
> 
> Rebooting...
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
> Saved PC:0x40376ed8
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3808,len:0x518
> load:0x403c9700,len:0xc08
> load:0x403cc700,len:0x2a0c
> entry 0x403c98dc
> [boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
> [   309][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(44) txPin(43)
> [   318][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
> [   326][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
> [   335][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not have default pins on ESP32S3!
> [  1575][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [  1704][I][partitioner.cpp:396] partitionCrawler(): [Partitioneer] Running partition is ESP_PARTITION_SUBTYPE_APP_TEST, no action taken
> [  1716][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x4202dee4 arg=0x3fc93d20
> [  1751][I][settings.cpp:327] getConfigs(): getConfigs: config.conf exists
> [  1777][I][settings.cpp:335] getConfigs(): getConfigs: deserialized correctly
> [  1787][I][settings.cpp:355] getConfigs(): Brightness: 100
> [  1792][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [  1803][I][settings.cpp:385] getConfigs(): Using config.conf setup file
> [  6868][I][main.cpp:91] get_partition_sizes(): label: vfs
> [  6873][I][main.cpp:102] get_partition_sizes(): [Partition Sizes] MAX_APP: 5111808
> [  6880][I][main.cpp:103] get_partition_sizes(): [Partition Sizes] MAX_SPIFFS: 1048576
> [  6888][I][main.cpp:104] get_partition_sizes(): [Partition Sizes] MAX_FAT_sys: 0
> [  6895][I][main.cpp:105] get_partition_sizes(): [Partition Sizes] MAX_FAT_vfs: 524288
> [  6931][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 255
> [ 15271][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 38 - Using Channel 7, Value = 169
> [ 15279][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 6 = 80 Mhz, APB: 80000000 Hz

auroJimen, 2024-08-16:
I don't now if maybe you've already read it, but i went searching on my own and found something that looks related (?):

https://stackoverflow.com/questions/66278271/task-watchdog-got-triggered-the-tasks-did-not-reset-the-watchdog-in-time

bmorcelli, 2024-08-17:
I followed this thread this time.. let me know if it works..

[Launcher_2.0_Cardputer3.zip](https://github.com/user-attachments/files/16642866/Launcher_2.0_Cardputer3.zip)

auroJimen, 2024-08-17:
The problem persists sadly.. :( Thanks anyway for putting all this time into it :)

> E (122838) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E (122838) task_wdt:  - async_tcp (CPU 0/1)
> E (122838) task_wdt: Tasks currently running:
> E (122838) task_wdt: CPU 0: IDLE0
> E (122838) task_wdt: CPU 1: loopTask
> E (122838) task_wdt: Aborting.
> 
> abort() was called at PC 0x42039210 on core 0
> 
> 
> Backtrace: 0x403772ca:0x3fc94e00 0x4037c1c5:0x3fc94e20 0x403820e5:0x3fc94e40 0x42039210:0x3fc94ec0 0x4037897d:0x3fc94ee0 0x420f42e7:0x3fcf4c20 0x42039a75:0x3fcf4c40 0x4037d76c:0x3fcf4c60
> 
> 
> 
> 
> ELF file SHA256: c2f59ef09a4dd110
> 
> Rebooting...

bmorcelli, 2024-08-29:
Hey buddy.. can you test this version, please? 
[Launcher_2.0_Cardputer.zip](https://github.com/user-attachments/files/16806372/Launcher_2.0_Cardputer.zip)

auroJimen, 2024-09-01:
Seems fixed to me :) Thanks a bunch

closed 2024-09-02 by bmorcelli