bmorcelli/Launcher #63: CardPuter Reboots During File Transfer from SD Card via Web Interface (M5Launcher Versions Above 2.0.1)

God-emperor-greatest, 2024-08-20:
### Description
I'm encountering a consistent issue with the M5Launcher on the CardPuter across all versions above 2.0.1. When attempting to download a file from the SD card on the CardPuter to my computer through the web interface, the device reboots a few seconds after the transfer begins. This problem occurs on multiple operating systems, including Windows, Linux, and Android.

### Steps to Reproduce
1. Insert a microSD card into the CardPuter.
2. Open the M5Launcher web interface on your computer or mobile device.
3. Attempt to download any file from the microSD card to the device.

### Expected Behavior
The file should download successfully without causing the CardPuter to reboot.

### Actual Behavior
The CardPuter reboots a few seconds after the file transfer begins, and the download is interrupted.

### Device Information
- **Device**: CardPuter
- **M5Launcher Version**: 2.0.1 and above
- **Operating System**: Windows, Linux, Android

### Logs and Screenshots
Unfortunately, I cannot capture any specific logs before the reboot, but I'm available to provide further information if necessary.

### Additional Information
The issue was not present in M5Launcher version 2.0.1 or earlier. It consistently occurs in all subsequent versions.

Thank you for looking into this matter!

bmorcelli, 2024-08-21:
It is probably related to AsyncTCP triggering the Watchdog timer, similar to https://github.com/bmorcelli/M5Stick-Launcher/issues/61

I'm studying how to disable/increase timers for the watchdogs in the framework.. maybe it solves the problem

bmorcelli, 2024-08-29:
Hey friend, can you test with this binary?
[Launcher_2.0_Cardputer.zip](https://github.com/user-attachments/files/16806407/Launcher_2.0_Cardputer.zip)

Install it using [This flasher](https://web.esphome.io)

God-emperor-greatest, 2024-08-30:
In this version, the reboot issue does not occur, but the file transfer is interrupted after 70 KB when tested with different files.

bmorcelli, 2024-08-30:
Well, the wifi tweak is not working for WebUI interface.. I am recompiling the framework and will recompile the project this night again.. I'll send a new binary in a few hours..

If it doesn't work, I will change from AsyncWebServer to the Webserver library.. it'll make the fw a little lighter, and will work the same way Bruce WebUI works

bmorcelli, 2024-08-30:
[Launcher_2.0_Cardputer.zip](https://github.com/user-attachments/files/16822505/Launcher_2.0_Cardputer.zip)

Lets see if it works..

God-emperor-greatest, 2024-09-01:
Same result

bmorcelli, 2024-09-02:
[Launcher_2.0_Cardputer_24-09-02.zip](https://github.com/user-attachments/files/16841350/Launcher_2.0_Cardputer_24-09-02.zip)

Now the framework is basically the same as the 2.0.1... lets give it a try

God-emperor-greatest, 2024-09-06:
Same result

closed 2024-09-10 by bmorcelli