bmorcelli/Launcher #66: Display stuck at various points??

raichea, 2024-08-28:
I've just got a Cardputer and have installed M5 Launcher on it, which is working well. 

After downloaded a binary to the SD card the screen appears to be stuck at "Downloaded". I randomly press buttons (ESC, Enter etc) hoping to clear it, but nothing does - eventually the display does return to normal operation. 

I see similar behaviour after listing paritioins... the list hangs around but can't be cleared although it does eventually go away.

is this intentional? If so, the delay does seem a bit long - I'd personally prefer that a suitable keypress cleared the displayed info.

bmorcelli, 2024-08-29:
> After downloaded a binary to the SD card the screen appears to be stuck at "Downloaded". I randomly press buttons (ESC, Enter etc) hoping to clear it, but nothing does - eventually the display does return to normal operation.

It is wainting for you to press "Ok" to go away 
 
> I see similar behaviour after listing paritioins... the list hangs around but can't be cleared although it does eventually go away.
> 
> is this intentional? If so, the delay does seem a bit long - I'd personally prefer that a suitable keypress cleared the displayed info.

It is waiting you press "Ok" to confirm the action, and go away

closed 2024-08-29 by bmorcelli

raichea, 2024-08-30:
When I said I pressed 'Enter' as one of the keys I tried, I should have said 'OK'. Portraying it doesn't remove the message.

bmorcelli, 2024-08-30:
it is waiting for 2s after press the btn https://github.com/bmorcelli/M5Stick-Launcher/blob/0ba9a86b23edb1822c293c4295d23d6bf7f2f54c/src/onlineLauncher.cpp#L276

I will remove it from the code as well, because it is now waitinh user´s input

raichea, 2024-08-31:
Ah, that explains it. Thanks.