bmorcelli/Launcher #67: Splash screen not readable

raichea, 2024-08-28:
Just an aesthetic thing, but when I first installed M5 Launcher, the splash screen showed the name before being obliterated by random characters. After using M5 Launcher to install some firmware, the splash screen is no longer readable - it's as though the first couple of seconds of the animation are being skipped.

bmorcelli, 2024-08-30:
Can you send me a video, please?

raichea, 2024-08-30:
No problem - here it is (it gets flipped during the upload for some reason).

https://github.com/user-attachments/assets/86df7d9a-c514-4c36-9458-f74cacde11ac

bmorcelli, 2024-08-30:
Ok, I will make the Launcher text with a different color (darker) in this green scheme..

raichea, 2024-08-31:
Thanks, that will improve things, but I'm wondering why the first part of the animation goes missing

closed 2024-09-10 by bmorcelli