bmorcelli/Launcher #68: Problem with m5launcher during OTA update

Teapot321, 2024-08-30:
Serial Monitor
[FROM M5Burner] COM3 opened.
[ 16270][I][display.cpp:545] loopOptions(): Number of options: 2
[ 19611][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 19615][W][sd_diskio.cpp:512] ff_sd_initialize(): sdWait fail ignored, card initialize continues
[ 19624][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[ 19631][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 20139][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 20143][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 20168][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 0 - WIFI_READY
[ 20250][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring SoftAP static IP: 172.0.0.1, MASK: 255.255.255.0, GW: 172.0.0.1
20250][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Started
[ 20267][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 10 - AP_START
[ 20275][V][WiFiGeneric.cpp:143] set_esp_interface_ip(): SoftAP: 172.0.0.1 | Gateway: 172.0.0.1 | DHCP Start: 0.0.0.0 | Netmask: 255.255.255.0
[ 20288][V][WiFiGeneric.cpp:190] set_esp_interface_ip(): DHCP Server Range: 172.0.0.2 to 172.0.0.12
IP: 172.0.0.1
Configuring Webserver ...
[ 21002][E][esp32-hal-misc.c:143] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT
[ 28549][V][WiFiGeneric.cpp:408] _arduino_event_cb(): AP Station Connected: MAC: 90:a2:5b:4f:8c:a3, AID: 1
[ 28559][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[ 29834][V][WiFiGeneric.cpp:422] _arduino_event_cb(): AP Station IP Assigned:172.0.0.2
[ 29842][D][WiFiGeneric.cpp:1040] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[ 39101][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 27 - Using Channel 15, Value = 193
[ 39694][W][AsyncTCP.cpp:950] _poll(): rx timeout 4
Listing files stored on SD
[ 48409][E][vfs_api.cpp:24] open(): File system is not mounted
[ 48417][E][vfs_api.cpp:24] open(): File system is not mounted
Folder: /
Folder: 
Folder: 
Folder: 
Folder: 
Folder: 
Folder: 
Folder: 
Folder: 
Folder: 
Folder: 
Folder: 
Folder: 
Folder: 
Folder: 
Folder: 
Folder: 
E (102915) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (102915) task_wdt:  - async_tcp (CPU 0/1)
E (102915) task_wdt: Tasks currently running:
E (102915) task_wdt: CPU 0: IDLE0
E (102915) task_wdt: CPU 1: loopTask
E (102915) task_wdt: Aborting.

abort() was called at PC 0x4010a484 on core 0

Backtrace: 0x40083a91:0x3ffbed0c |<-CORRUPTED

ELF file SHA256: 1933d9d5130617ca

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1276
ho 0 tail 12 room 4
load:0x40078000,len:13232
load:0x40080400,len:3076
entry 0x400805f0
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
E ���68) esp_ota_ops: not found otadata
[    41][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   277][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   315][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[   324][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[   335][V][esp32-hal-uart.c:463] uartBegin(): UART0 initialization done.
[   350][I][M5GFX.cpp:532] init_impl(): [M5GFX] [Autodetect] load from NVS : board:5
[   361][D][M5GFX.cpp:494] _read_panel_id(): [M5GFX] [Autodetect] read cmd:04 = 00528585
[   369][I][M5GFX.cpp:750] autodetect(): [M5GFX] [Autodetect] M5StickCPlus2
[   507][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[   546][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(-1) txPin(-1)
[   555][V][esp32-hal-uart.c:343] uartBegin(): UART0 Driver already installed.
[   564][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=21 scl=22 freq=100000
[  1837][W][EEPROM.cpp:81] begin(): truncating EEPROM from 128 to 64
[  1851][V][esp32-hal-ledc.c:231] analogWrite(): GPIO 27 - Using Channel 15, Value = 255

bmorcelli, 2024-08-30:
Relate to https://github.com/bmorcelli/M5Stick-Launcher/issues/61 the Watchdog timers being triggered by AsyncTCP..

Please try with this binary https://github.com/user-attachments/files/16806407/Launcher_2.0_Cardputer.zip

bmorcelli, 2024-08-30:
[Launcher_2.0_Cardputer.zip](https://github.com/user-attachments/files/16822507/Launcher_2.0_Cardputer.zip)

try this one, please

bmorcelli, 2024-09-02:
[Launcher_2.0_Cardputer_24-09-02.zip](https://github.com/user-attachments/files/16841354/Launcher_2.0_Cardputer_24-09-02.zip)

Can You test this one?

Teapot321, 2024-09-07:
I have plus2

closed 2024-09-10 by bmorcelli