bmorcelli/Launcher #75: Port for T-Dongle-S3

carlosfd1, 2024-10-13:
Hallo,
do you know if the Launcher ist compatible with the T-Dongle-S3?
(https://www.lilygo.cc/products/t-dongle-s3)
It uses a ST7735 display and has a ESP32-S3 Xtensa LX7 as MCU.

bmorcelli, 2024-10-13:
Not yet...

I'm planning a new version where you'll be able to use the Launcher on "any" esp32 (and s3).. by launching the Web User Interface..

It would "solve" the UI porting issue and the necessity of at least 3 btns to use the Launcher..

I believe I will release this version in a few days, and I'll tag this issue on that, and this other too 
https://github.com/bmorcelli/M5Stick-Launcher/issues/51

carlosfd1, 2024-10-13:
Thank you for this fantastic answer! I'm really looking forward for the new release.
Best regards!

bmorcelli, 2024-11-01:
I've made a "Headless" ESP32-S3 version that might be useful for this device...

You will flash the option "[Espressif - ESP32-S3-16mb](https://bmorcelli.github.io/M5Stick-Launcher/) in the web flasher
- After first flash, it'll be asking for restart, on serial Monitor, so unplug and plug again from USB
- Now, it will turn On, and if you press the GPIO 0 btn (boot btn ) within the first 5s, it will enter the launcher, otherwise it'll go to the previously installed firmware (by Launcher)..
- if you can, use a serial Monitor Application to follow the instructions
- the first time, it will star the WebUi in access point mode, so turn off mobile data and connect to Launcher wifi access point.
- access http://172.0.0.1 or http://launcher.local in your browser
- when the page opens, login is "admin", pwd "launcher"
- in the web user interface (WebUI), you can:
    -  configure the SD pins to read the SDCard (need restart to work)
    - configure your network to access the launcher from your computer in your home network (need restart to work)
    - manage your SDCard, and 
    - install firmware using OTA Update option, using binary files compatible with your device