bmorcelli/Launcher #76: can't insall microPython firmware on cardputer

Yixi, 2024-10-23:
Installing the MicroPython firmware on the cardputer gets stuck at the final ‘complete’ stage and won’t boot. The firmware can be downloaded from: https://micropython.org/download/ESP32_GENERIC_S3/ 1.23.0

bmorcelli, 2024-10-23:
Micropython firmware, by default, doesn't have a UI that sends commands to the tft, so the tft keeps "frozen" in the last state (the last command it received) while the firmware is running in the MCU.

Start the Launcher, plug it to the Arduino IDE Serial Monitor, and install this binary... You will see a few installing messages, and after that a reboot command and then confirm that the Micropython boot messages are shown after that.

If it keeps in boot loop, please inform me.

Yixi, 2024-10-23:
After restarting, I found that the MicroPython environment had indeed loaded. Previously, connecting to Thonny kept giving a ‘Resource busy’ error, but now it works fine. However, when debugging with MicroPython based on M5Launcher, it’s more likely to encounter a MemoryError.

 By the way, thanks for the response.

closed 2024-10-23 by Yixi