bmorcelli/Launcher #77: M5 cardputer launch UIFlow2 reboots in a loop

Yixi, 2024-10-24:
Already change the partition scheme to "UIFLOW2"

M5Launcher version 2.3.0
UIFlow version 2.1.6

download from the OTA and install, but can't lanch, reboots in a loop

anybody can help me?

bmorcelli, 2024-10-24:
Hello my friend,

thank you for the Report!

UIFlow Application became bigger than what Launcher can handle in this last update... Launcher reserves an area of 0x450000 bytes for it ( https://github.com/bmorcelli/M5Stick-Launcher/blob/34b126141dd7962053e0df7663f14931051c8139/custom_8Mb.csv#L21 ), and in a quick analysis UIFlow factory application increased to 0x45B190 bytes.. So I need to adjust it to at least 0x460000 bytes

So I need to remake the partition scheme for this, and I´ll release the next version on M5Burner in a few hours.

bmorcelli, 2024-10-24:
Solved in this release 2.3.1, use the webflasher.. i will upload to m5burner in a few hours

closed 2024-10-24 by bmorcelli