bmorcelli/Launcher #78: UIFlow 2.0 not starting after installation

dobisan, 2024-10-30:
I installed UIFlow 2.0 throught OTA but it doesn't start.

I tried to change the partition to UIFlow2.0 and DOOM as well but UIFlow 2.0 (2.1.5-hotfix) doesn't start. It's just a black screen.

Tried to delete the FAT before. Doesn't work either.

Maybe I doing something wrong but I don't know what.

dobisan, 2024-10-30:
Sorry I doesn't mentioned the M5 launcher version I use and the device.

I use M5 launcher 2.3.1 on an M5Stack Cardputer.

I noticed in the linked issue UIFlow2.0 version 2.1.6 is mentioned.

There is a 2.1.6-hotfix version. On the cardputer in the OTA option only that version are available.

I also tried to flash these versions through the webui OTA update function:
2.1.6-hotfix
2.1.6
2.1.5

Still a black screen.

I have to set the partition I want to use before every firmware installing attempt?

I have to clear the FAT before flash a big firmware like uiflow?

bmorcelli, 2024-10-30:
I will take a look over this hotfixed uiflow to see if it increased again to fix the partition scheme.

If you want to install UiFlow2, you have to set the partition scheme to UiFlow2 and restart the device.. after that it should be armed to install

dobisan, 2024-10-30:
Thanks for the reply.

A strange addition to the issue. The device is not boot-loop in this case. The screen just blink once after Launcher starting screen and remains black.

closed 2024-12-06 by bmorcelli