bmorcelli/Launcher #80: CYD support (2.4+Capactive)

Teapot321, 2024-11-01:
I would like you to add support for the new CYD.
I have this CYD and I can't figure out what it is.
It has a 2.4screen as well as a Capactive touch, ESP32-2432S024 is written on the back, this CYD has a Type-C port. It has pins (GND, O35, O22, O21) x2 as well as Bat+ Bat- and one I2C. The packaging said ESP32-2432S024C

bmorcelli, 2024-11-25:
Hello...

Can you please try the beta version?

[Web Flasher](https://bmorcelli.github.io/M5Stick-Launcher/)
[image]

bmorcelli, 2024-11-25:
Great!! Thank you!

closed 2024-11-25 by bmorcelli

Blacktrecks, 2025-01-25:
I would like you to add support for the new CYD.
I have this CYD and I can't figure out what it is.
It has a 2.4screen as well as a Resistive touch, ESP32-2432S024 is written on the back, this CYD has a Type-C port. It has pins (GND, O35, O22, O21) x2 as well as Bat+ Bat- and one I2C. The packaging said ESP32-2432S024C. I have tried all the versions and release it only sbow me the interface for the capacitive variant... I dont know what to do