bmorcelli/Launcher #81: Stuck in firmware download menu

MrAureliusR, 2024-11-08:
I just updated to v2.3.1 and I noticed quickly that there is no longer a way to exit back to the firmware selection menu once you've selected a firmware to download. The Esc key does nothing if there is only one version of that firmware available, or it scrolls in reverse order through the versions.
This forces you to select "Main Menu" from the options menu which disconnects you from WiFi and boots you back out to the main M5Launcher screen. Before, hitting escape would simply take you back to the firmware selection menu and allow you to browse other firmware without having to re-enter the OTA mode.

I hope this all makes sense, I suspect it's probably just a minor regression.

bmorcelli, 2024-11-08:
Thank you for your report..

I'll check it soon

bmorcelli, 2024-11-09:
was looking into the code now, and I added a logic for StickCs and other devices where we need to hold the "previous" btn for one second..

Try holding Esc ( ` ) for some time, I think it'll work this way..

I will fix it, but this is how to get out for now

closed 2024-11-25 by bmorcelli