bmorcelli/Launcher #82: Enhancement: going back to firmware list in OTA menu

shurillu, 2024-11-09:
Hello Bernardo,

   thanks for your great job! Could you implement the option to going back to the firmware list in OTA menu? 
Actually when you select OTA, you have to select an AP (and enter the password, for the first time you use that AP) and then the cardputer download the firmware list.
Once you select one firmware, you can't go back to the firmware list, but you have to going back to the main menu.
So, if you want to install several firmwares from OTA you have to do all the long way: OTA -> AP -> Download firmware list -> select firmware -> install (on SD in my case) -> Main Menu -> repeat.

Could be a great upgrade going back to the firmware list at the firmware position previously selected.
The SD menu works in a similar way that I'm pointing out by pressing ESC, but it reload the list in spite of storing it and going back to the previous position.

Do you think it is possible to implement?

Thanks a lot, regards.

Stefano

bmorcelli, 2024-11-09:
What's up friend, it is possible, yes..

In fact, this is the desired action that stopped working in this last update, when I ported T-Embed 

I will fix this soon, just need to finish a complex task on Bruce Fw

bmorcelli, 2024-11-09:
I was looking into the code now, and I added a logic for StickCs and other devices where we need to hold the "previous" btn for one second..

Try holding Esc ( ` ) for some time, I think it'll work this way..

shurillu, 2024-11-09:
Thanks for the very quick fix.
I'll try it.

bmorcelli, 2024-11-25:
Can you test the new Beta Version of the Launcher, where I have added ways to exit from OTA List and from Wrong passwords on Wifi??

Exiting from Keyboard itself is not available yet (And sincerily don't know if it will be)
[Web Flasher](https://bmorcelli.github.io/M5Stick-Launcher/)
[image]

shurillu, 2024-11-25:
Hello Bernardo, 
   yes of course, I have some M5Stack product to use as a test equipment. I'll give you a feedback asap.

Stefano

shurillu, 2024-11-26:
Good! Tonight I'll check the new beta and I'll give to you some feedback!

shurillu, 2024-11-26:
Hello Bernardo,
I've installed the beta version 630460e on my Cardputer. Here what I found.
- From OTA, when the Access Point List is displayed, ESC single pressed go back to the main menu
- From OTA, It is possible to go back to the main menu (SD - OTA - WUI etc) pressing **two times** ESC key (same with backspace) from any point of the program. A Long press do the same (go back to the main menu).
- From OTA, when a firmware is selected, selecting the "Options", the entry "Back to the list" don't go back to the list but remain in the firmware detail. You have to push two times ENTER to select "Back to the list". To select the other entries (OTA Install, Download->SD) a single ENTER press is suffice.
- From SD menu, it is possible to go back to the main menu (SD - OTA - WUI etc) pressing **once** ESC key (same with backspace) from any point of the program.

Core 2 (same version, 630460e)
- Now the installer work properly
- The buttons (PREV, SEL, NEXT) are a bit cut in the bottom (like the screen is "slided" to the bottom)
- in the OTA menu, the PREV button navigate backward in the firmware list. A long press did nothing
- From OTA in the Access Point selection, PREV long pressed go back to main menu
- Once a firmware is selected, PREV long pressed go back to the list
- There is no way to go back to the main menu from OTA 
- On SD menu, there is the "back" entry to go back, so it work good

Hoping it helps, regards.

Stefano

bmorcelli, 2024-11-27:
Great..
>The buttons (PREV, SEL, NEXT) are a bit cut in the bottom (like the screen is "slided" to the bottom)

It is intentional, I'm using the same codebase from bruce to keep things simple..

I just added Scrolling text on filelist (SD) and SSID network listings..

thank you for your help!

shurillu, 2024-11-27:
> I just added Scrolling text on filelist (SD) and SSID network listings..

Yep, I saw it but I forgot to mention it. Nice work!

closed 2024-12-06 by bmorcelli