bmorcelli/Launcher #83: Maurader Mini tactile button

Drotnip, 2024-11-09:
Hello Sir. I was wanting to ask how is the button programed. What does what. I cannot enter my wifi password correctly. It has skipped places and jumps 2 spaces. Flashed my esp 32 marauder mini I enjoy the options but not that one.

bmorcelli, 2024-11-13:
Hey friend, sorry for the delay..

It is actually configured to use as 3 btns interface

Moving leveler UP: 
* Short pulse: move the cursor down
* Long pulse: move the cursor up

Moving cursor Down: 
* Short pulse: move cursor Right 
* Long pulse: move cursor Left

Center click: Select

It is very counterintuitive, I understand, but I will fix it in the next release ;)

To help you now, Launcher creates a file in the SD Card, called "config.conf".... Open it in the notepad, and change the MyWifiSSID and MyWifiPwd (or something like that) with your values... As it is a json, if you follow the structure you can add more networks i to it.. when connecting to a known network it will use this value

Drotnip, 2024-11-13:
Thank you very much. There is no need to apologize for any delay. You are a busy man in a demanding field. I'll do what you said and change it. Muchas Gracias.

bmorcelli, 2024-11-25:
I'm finishing the new interface scheme for new boards, and tried to add side navigation for marauder mini devices.

Can you, please, test the Beta Release, to make sure that this build is ok? [Web Flasher](https://bmorcelli.github.io/M5Stick-Launcher/)

[image]

Drotnip, 2024-11-25:
After clicking Beta version then marauder then mini it refreshes page. no
option to connect or install

bmorcelli, 2024-11-25:
don't click the yellow letters lol.. I need to fix it.. hit the box

Drotnip, 2024-11-25:
Got it. I'm in

bmorcelli, 2024-11-25:
> don't click the yellow letters lol.. I need to fix it.. hit the box

Fixed the btns.. now links are displayed only when available

closed 2024-11-25 by bmorcelli

bmorcelli, 2024-11-28:
Can you test the Marauder mini version from Beta Release? It is supposed to use Left and right for navigation..

If it keeps rebooting after show the bootscreen, keep the center btn pressed until it get into the Launcher, and tell me to what side the menu is going