bmorcelli/Launcher #86: SD Card Issues on 2.4.0 on T-deck Plus

DOPEVAPOREON, 2024-12-06:
SD Card is not detected on 2.4.0 on my T-deck Plus.  This issue is not present in 2.3.3, and SD card functions as normal on this version.  Let me know if there's any more info I need to provide.

Ty-Stelow, 2024-12-06:
I am also having an issues with the SD card not being detected on 2.4.0 with T-Deck Plus.

bmorcelli, 2024-12-06:
Hello!! I think this issue was occuring due to SPI bus jamming caused by LoRa chip not being initialized (disabled by launcher startup), so it causes interference in the SD Card communication.

I think it is supposed to work now, can You test installing the BetaRelease Version??

https://bmorcelli.github.io/M5Stick-Launcher/

Coose BetaRelease at the webflasher, and tell me if it works for me to release the patch officially ;-)

closed 2024-12-06 by bmorcelli

tiuksferve, 2024-12-19:
Hello bmorcelli.
Is the web flasher not working?
There is no connect option after putting the device on download mode and selecting the desired version.