bmorcelli/Launcher #87: Is there any way to save the modified flash back to the SD card before reflashing?

jstockdale, 2024-12-08:
I'm wondering if there's any way to persist the changes that have been made by the device. For instance I lose Meshtastic settings everytime I boot a different firmware.

Happy to help implement if I can be useful! :)

bmorcelli, 2024-12-09:
You can... I didn't analyse the meshtastic firmware, but it would probably use SPIFFS to save data ..

You can go: CFG>Save SPIFFS

It will save a file in ./bkp/spiffs.bin

Then you can restore it in CFG>Rest SPIFFS> choose the file you have saved and select.

jstockdale, 2024-12-10:
Oh my goodness. That worked!! If I restore the flash first, then do Rest SPIFFS with the saved file, my configuration and settings are all restored.

If I put together a PR that lets you set configuration preferences for certain bins, so that you can auto-save and auto-restore the SPIFFS or FAT VFS data, would you be interested in the feature?

Awesome project. Thanks for making it! 👍 🙏 ✨

jstockdale, 2024-12-14:
Sounds great. I will get to this later this month. Cheers! 😊✨

mmiller7, 2025-02-28:
> You can... I didn't analyse the meshtastic firmware, but it would probably use SPIFFS to save data ..
> 
> You can go: CFG>Save SPIFFS
> 
> It will save a file in ./bkp/spiffs.bin
> 
> Then you can restore it in CFG>Rest SPIFFS> choose the file you have saved and select.

Do you mean "Restore SPIFFS"?

Any way to verify the backup without risking losing data?  I had less success in my first try seemed to not restore it.

jstockdale, 2025-02-28:
Also sorry I've been busy. Let me look into making this feature addition
soon!

closed 2025-10-28 by bmorcelli