bmorcelli/Launcher #90: Support for LilyGo FabGL VGA32 ESP32

leorrr, 2024-12-13:
Hello, first of all, congratulations on your work! I wanted to ask if it would be possible to add support for the LilyGo FabGL VGA32 ESP32 PSRAM Module V1.4 Controller (https://lilygo.cc/products/fabgl-vga32?srsltid=AfmBOoqFQGGNJfciDUQ4AnBUxMFLPkxPOaW4LMzcUfwpUfx-Tcuz1t6z). This board is used to run emulators of classic computers like the Spectrum, VIC-20, etc., and it would be very interesting to switch between them without having to reflash the board using a PC. 
Best regards.

bmorcelli, 2024-12-13:
The Headless version of the Launcher might do the trick.

Instructions are in this issue 😉

https://github.com/bmorcelli/M5Stick-Launcher/issues/51#issuecomment-2452680385

leorrr, 2024-12-13:
Okay, thank you. I just saw it before reading your response. I’ll see what can be done. Thanks again, best regards.

closed 2024-12-13 by leorrr