bmorcelli/Launcher #91: Any attempt to install firmware from SD card returns "update error."

LucidEye, 2024-12-23:
Using resistiv touch-screen CYD-2432S028 model with correct firmware. (v2.4.1 Launcher-CYD-2432S028.bin)
Flashed using the online "Launcher Flasher" and firmware loaded OK and appears to work, at first.... can browse through menus and change settings OK.

Downloaded several different (known working) firmware (.bin files) for testing to a fat32 formatted SD card.
.bin files are specifically for CYD-2432S028
Firmware was sorted into folders for organization.

I can use the touch-screen menu to see the SD card and all the folders and files stored on it.

However, every firmware I try to install throws the error "update error." and nothing happens.

I have tried erasing the CYD and re-flashing the Launcher twice and still have same issue.

What could I be doing wrong?

bmorcelli, 2024-12-24:
I helped 2 people today with similar problems, and turns out it was a problem with the SD Card format..

I always used SDHC cards lesser or equal to 32gb formatted in FAT32 using the windows formatter.. I have used Rufus once, and it worked well too..

LucidEye, 2024-12-24:
I'm using a name-brand 32gb SDHC formatted in GParted with an msdos partition table and FAT32.
In fact I've tried 2 of them, but they are the same exact card and I did format them both in GParted.
Like I said, In the Launcher firmware, I can navigate to the card and see everything on the card but I just can't install any of the firmware... weird.  
I haven't touched a Windows machine in probably over 10 years at this point... strictly Linux here ;-)
I don't even think I have any old Windoze VMs I can spin up anymore either :-P
Odd, never encountered any problems using SD cards formatted in GParted on any of my IoT devices.... it's been my go-to formatting and partitioning tool for years.
I'll guess I'll just try to see if I have any other cards or formatting tools and let you know what happens... I think I only have one SD card under 32GB... need to remember where I put it :-P
Thanks for the quick reply... and great work on the CYD version of the Launcher... looks like a really useful tool for rapid testing of firmware!

LucidEye, 2024-12-24:
Wow... actually found a crusty-old 8gb card and that seems to be working... firmware is installing now!
Thank you again.

closed 2024-12-30 by bmorcelli