bmorcelli/Launcher #92: M5 cardputer launch UIFlow2 reboots in a loop (Was Issue #77)

wooodz, 2024-12-23:
Hi,

I am having the same problem as reported in issue #77 Using version UIFlow2.0 (v2.2)

and Launcher v2.4.1 where Launcher keeps rebooting trying to load UIFlow.

wooodz, 2024-12-23:
Also to add if I try to install UIFlow2.0 (2.1.8 or 2.1.9) I just get a blank screen after launcher loads. I have set the CFG to UIFLOW2 and when I list the part I get OTA-FAT-FAT.

bmorcelli, 2024-12-24:
Are you installing through OTA or with the binary in your SD Card?

wooodz, 2024-12-24:
Hi 

Also just tried OTA update it installed FW and connected and updated FAT twice. On restarting it just reboots over and over again.

bmorcelli, 2024-12-30:
UIFlow is getting bigger and bigger.. I think I fixed it in the Beta Release, that you can install using the web flasher

https://bmorcelli.github.io/M5Stick-Launcher/

closed 2024-12-30 by bmorcelli

wooodz, 2024-12-30:
Thanks all working now.