bmorcelli/Launcher #93: M5 laucher doesn`t work on CYD JC2432W328C

doniomm, 2024-12-28:
Hello, 

I flashed bruce final to the CYD JC2432W328C and it worked fine :-)

but the m5launcher doesn`t work on it :-( 
[image]
[image]

the screen is always black.

resolution 240*320
W3-2.8 inch Caspacitive touch

```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1276
ho 0 tail 12 room 4
load:0x40078000,len:13232
load:0x40080400,len:3076
entry 0x400805f0
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 3
ets Jul 29 2019 12:21:46
```

bmorcelli, 2024-12-30:
I think I have it fixed now, can you test in the Beta Release, please? 

https://bmorcelli.github.io/M5Stick-Launcher/

doniomm, 2024-12-30:
now it works, thx for your support
I love it

closed 2024-12-30 by doniomm

Blacktrecks, 2025-01-25:
I  have the same problem on 2.4 and i tried even the beta, when i switch to capactive version it seems to work the interface vut my display is rezistive