bmorcelli/Launcher #95: Installation Failed OTA Updates [StackCPlus2]

slgstudio, 2024-12-30:
OTA firmware install always fails. I've tried many different networks and hotspots and I've tried every version of the M5Launcher firmware. Doesn't work at all. Works perfectly on cardputer. It just says "Installation Failed" instantly after initiating the download.

bmorcelli, 2025-01-05:
> I am having the same issue. Did you manage to fix it?

Not yet.. It is working on my side 😕

bmorcelli, 2025-01-05:
Json file (database) is too big.. doesn't have enaugh space for the certificate..

m5burber has too many trash fw and dummy copies (many Nemos, Bruces and Marauders).. Ill try to fix it on launcher side

bmorcelli, 2025-01-05:
Can you test the Beta release, pleas?

https://bmorcelli.github.io/M5Stick-Launcher/

slgstudio, 2025-01-05:
Wont let me flash it :/ I put it into download mode and says "Failed to
execute 'open' on 'SerialPort': Failed to open serial port."

dylancolwill, 2025-01-06:
Just flashed the beta using the online tool, then Bruce using M5Burner database. Haven't tested using the web server but otherwise working so far. Thanks 👍

dylancolwill, 2025-01-06:
> Wont let me flash it :/ I put it into download mode and says "Failed to execute 'open' on 'SerialPort': Failed to open serial port."

I had this error the first time. Flashed successfully when putting the CPlus2 into download mode by connecting ground and g0 pins.

closed 2025-01-07 by bmorcelli