bmorcelli/Launcher #98: M5Core basic button error

Erwantest123, 2025-01-07:
Hello @bmorcelli,
====
I wanted to let you know that I tried the new version on my M5Core Basic. However, when I pressed the B button to enter the Launcher, it didn't detect the button press.
Looking forward to your response.
Best regards,
@Erwantest123

bmorcelli, 2025-01-07:
This is weird.. I will look into this

bmorcelli, 2025-01-08:
One question... Once you have installed, it goes to the Launcher screen.. there the buttons work as intended or not?

Erwantest123, 2025-01-08:
no the buttons don't work and yes it goes to the launcher it acts as if I was waiting without pressing a button while I press. @Erwantest123 Google translate

50u15pec7a70r, 2025-01-11:
Same problem here. 
v.2.4.2 has button issue. 
v.2.4.0 works fine.

bmorcelli, 2025-01-11:
hey guys.. Think I found the problem.. 

I have a patch to test on BetaRelease, you can fid it here https://bmorcelli.github.io/M5Stick-Launcher/

please lmk if it worked as expected

50u15pec7a70r, 2025-01-12:
> hey guys.. Think I found the problem..
> 
> I have a patch to test on BetaRelease, you can fid it here https://bmorcelli.github.io/M5Stick-Launcher/
> 
> please lmk if it worked as expected

I don't use Google Chrome or Microsoft Edge. Sorry.

closed 2025-01-13 by bmorcelli