bmorcelli/Launcher #101: Launcher 2.4.2 non si riavvia con Lilygo T-embed

Luigi-77, 2025-01-08:
Scaricando la nuova versione di Launcher 2.4.2 su Lilygo T-embed e avviando il dispositivo tutto sembra funzionare nel momento in cui si sceglie un nuovo firmware dalla scheda SD e parte il programma scelto quando viene spento e poi riacceso in un secondo momento Launcher non c'è più riparte subito l'ultimo programma installato senza possibilità di sceglierne un altro grazie per l'attenzione.

bmorcelli, 2025-01-11:
Quando si riavvia, viene visualizzata la schermata verde del Launcher? Se appare, premi il pulsante Encoder per accedere al launcher. Se non lo premi, salterai il passaggio diretto all'altro FW, come previsto.

Luigi-77, 2025-01-11:
No la schermata verde del Launcher  non appare più riparte dall'ultimo firmware installato per far comparire Launcher devo togliere e rimettere la batteria cosa che prima non accadeva.

bmorcelli, 2025-01-11:
You're right.. it was using tge wrong framework in this version 

Can you try the BetaRelease, please?

Luigi-77, 2025-01-12:
ok adesso provo

Luigi-77, 2025-01-12:
ok tutto funziona correttamente con la versione beta, grazie.

closed 2025-01-13 by bmorcelli