bmorcelli/Launcher #99: Support ESP32-2432S028R " Resistive Touch USB Type-C Only

Ruslan228asr, 2025-01-08:
Good day! My name is Ruslan, and I'm writing from Ukraine. Recently, I encountered a problem with the firmware for the CYD-2432S028R (Type-C). The standard CYD-2432S028 firmware was installed on it, but there was no image. However, the CYD-2432W328C firmware displayed the image, but the touch screen didn't work.

I tried installing all the firmware versions from the official website, but nothing worked. The display itself is functional, as I've used it before. I also confirmed its operability with the Marauder firmware, where everything works fine.

I hope you can add support for this board. Many users on forums and Discord channels have reported similar issues with this board.

ShocKwav3, 2025-01-18:
I have the same issue with this variant

class5vapor, 2025-01-22:
I'm having the same issue, I can get Launcher to run and work, and the touch works great in it ... but when installing any firmwares ... the touch does not work in them...
Any ideas / fix ?

Mine is the 2.4 inch screen USB-C only model, same as OP, ESP32-2432S024R

bmorcelli, 2025-01-22:
I couldn't make touchscreen work in this device yet.

Touch driver shares the SPI bus with the tft, but when it runs, the watchdog is restarting the firmware on Bruce (where I'm doing the tests).. Launcher framework has all watchdogs disabled, so it can be the difference..

If you got it to work, tell me the secret.. 

But this issue of other firmware doesn't work the touchscreen with, it is kinda impossible, because when the launched firmware runs, all code from Launcher keeps inactive, it usually is problem with the other firmware..

bmorcelli, 2025-02-16:
Hey bro, can you test it for me?

Available on the [WebFlasher, BetRelease](https://bmorcelli.github.io/M5Stick-Launcher/) > CYD 2432W328R or 2432S024R (Resistive)

class5vapor, 2025-02-16:
Sure Sure...gimme a bit, watchin' hockey (I'm Canadian, I HAVE-TO! LoL)

🙂

class5vapor, 2025-02-16:
Doesn't seem to work, I mean...it installs... it's inverted color and touch
doesn't work

class5vapor, 2025-02-16:
OK, I got it to work, thank you...but it was under THIS selection that I
got it to work for me with NOT inverted colors and touch DOES work.
See the Pic

To re-cap, mine's the 024 or 2.4 inch usb-c only model.
Model: ESP32-2432S024

=]

Thank You Thank You

=]

[image: image.png]

class5vapor, 2025-02-16:
OK, so...one-thing...yes the Launcher works and the touch works..BUT if i
install any firmwarese ( I have Bruce, Ghost and Marauder) ... none of them
respond to touch ... yet touch DOES work on the initial SKIP / LAUNCHER /
SKIP screen ...
so...it's not really working still

Sorry sir / ma'am ... I hate to report bad news but ... you probably DO
want to know the truth.

=]

class5vapor, 2025-02-16:
I'm suspecting that it's probably something on MY side tho.
The Launcher WILL work with touch and all ... I'm thinking it's the
firmwares I'm installing - they MAY be the wrong ones...you see I have a
2.8 inch screen one too and these all work on it BUT ... I realize now ...
I've been trying to install these on the 2.4 inch screen on (the one I'm
testing here) and of-course those shouldn't work .. because they're for a
different CYD .. .sooooo ... yeah sorry about that...you can probably
ignore all the above, tomorrow I'll download the 2.4 inch screen firmwares
and try those to test ...

Sorry man, I get all mixed up sometimes...

=]

bmorcelli, 2025-02-20:
Yeah, you need to flash compatible versions to your device.. otherwise they won't work 

I'll close this issue since the port is done 😜

closed 2025-02-20 by bmorcelli