bmorcelli/Launcher #103: Feature request. More integrated Meshtastic and Ripple support

TBNRregs, 2025-01-24:
Heyo, I been using your launcher for some time on Lilygo T-Deck for switching between the normal and fancy Meshtastic FW, now recently started doing Ripple (Meshcore) and it's been working great until now and noticed a bit of a info save issue and secondly the newest 2.5.20 Fancy UI FW doesn't work with the launcher, but standalone is fine. 

I was wondering if it would be possible to enhance the integration of your launcher to have a better firmware switching experience.

bmorcelli, 2025-01-24:
I need the binarie files from these firmware to analyze and figure out how to fix it

Probably it is a partition scheme issue

TBNRregs, 2025-01-24:
[FW.zip](https://github.com/user-attachments/files/18540934/FW.zip)

Ripple one is gotten from the [here](https://buymeacoffee.com/ripplebiz)
Meshtastic 2.5.18 from Meshtastic github
Fancy UI 2.5.17 from gitlab build from this [video ](https://www.youtube.com/watch?v=LBJVCYypIG0) with a added modification as seen in the pinned comment
Fancy UI 2.5.20 build by recrof from Meshtastic discord

bmorcelli, 2025-06-05:
These firmwares (meshtastic and MeshCire) are available for OTA install in the newer versions of launcher (above 2.4.9)

closed 2025-06-05 by TBNRregs