bmorcelli/Launcher #108: Question: Porting to other devices

Shadowtrance, 2025-02-06:
I was looking to hopefully port this to a couple of other screens i have here. But they aren't supported in TFT_eSPI which from what i can see is the only display driver library being used.
They do however all work with [Arduino_GFX](https://github.com/moononournation/Arduino_GFX) and the last 2 also work with [LovyanGFX](https://github.com/lovyan03/LovyanGFX). Both libraries work with a heap of display (and touch in regards to LovyanGFX) drivers.
But yeah not really sure where to start so if you have any tips on how to go about porting them to launcher that would be great. :)

| Model           	| Driver    	| Touch     	| Resolution 	| Size     	| Type      	|
|-----------------	|-----------	|-----------	|------------	|----------	|-----------	|
| JC3248W535C     	| AXS15231B 	| AXS15231B 	| 320x480    	| 3.5 inch 	| QSPI      	|
| ESP32-8048S043C 	| ST7262    	| GT911     	| 800x480    	| 4.3 inch 	| RGB Panel 	|
| JC8048W550C     	| ST7262    	| GT911     	| 800x480    	| 5 inch   	| RGB Panel 	|

bmorcelli, 2025-02-06:
Hey mate, I'm glad to hear that..

I was looking into ArduinoGFX library yesterday and thinking of a way to migrate from TFT_eSPI to this library..

I will try something out today, and check if I can get it to properly work with the existing borders.. 

If it works, we can easily port to the boards you listed

T-REX-XP, 2025-02-06:
Hi guys. I have bought the CYB with type c: esp32-2432s022c . It seems like it has a 2.2-inch capacitor display. Any chance to add support of that board as well? I can provide any details. Thanks

```
Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
Connecting...
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: Wi-Fi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:15:18:d4:59:d4
```

[image]

[image]

Shadowtrance, 2025-02-06:
> Hey mate, I'm glad to hear that..
>
> I was looking into ArduinoGFX library yesterday and thinking of a way to migrate from TFT_eSPI to this library..
>
> I will try something out today, and check if I [...]

Hopefully it isn't too much of a pain to get working as I think a lot of things are similar between the two, Arduino GFX just supports a heap more displays).
In regards to touch, [bb_captouch](https://github.com/bitbank2/bb_captouch) covers a lot of common touch controllers and works pretty well imo.
Look forward to hearing how things go 😀

bmorcelli, 2025-02-25:
I am preparing to port to many of these Sunton boards (Cyd variants), so I'm changing a few things to better handle the tons of environment.. 

It is mostly missing the touchscreen for the devices with GT911 and AXS15231B

I hope I'll have it until the end end of the week.. so we will need to run the tests

Shadowtrance, 2025-02-25:
Crazy! If you need me to test anything with my boards (if you don't have them) let me know.

bmorcelli, 2025-02-26:
3 more devices to test CYD-2432S032C , CYD-8048S043C, CYD-8048W550C (If this works, Launcher will have more 20 boards in the list)

@T-REX-XP it is a good momment to retest the 2432S022C again.. I 

JC3248W535C -> For this guy I need the pins used for touchscreen (SDA, SCL) 

[image]

Shadowtrance, 2025-02-27:
> JC3248W535C -> For this guy I need the pins used for touchscreen (SDA, SCL)
SDA = 4
SCL  = 8
INT = 3
RST = -1

bmorcelli, 2025-02-27:
Ok.. you can try it again.. it was a flasher wrong setup

Shadowtrance, 2025-02-27:
E (388) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0 on both boards.

bmorcelli, 2025-02-28:
Apparently this error is related to QIO flash mode https://github.com/HomeSpan/HomeSpan/discussions/490

I changed to DIO flash mode to test.. it is interesting, because most ESP32-S3 use QIO flash mode

Shadowtrance, 2025-02-28:
yeah that's kinda weird. and i can confirm these boards do indeed use QIO flash mode. I even have "memory_type": "qio_opi" and "flash_mode": "qio" in my board json files for when i use platformio.

bmorcelli, 2025-02-28:
yes.. the json files from Sunton repo have it.. but worth a try.. tell me if it works

Shadowtrance, 2025-02-28:
I tried it and same result.

bmorcelli, 2025-03-02:
I changed the pins_arduino.h file, pointing to the right file.. can you test it on  CYD-8048S043C, CYD-8048W550C ??

Shadowtrance, 2025-03-02:
Tried the latest beta build after a full flash erase (just to have a clean slate) and still hitting that core dump flash config error and boot loop (nothing on screen)
```E (387) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377bd0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x518
load:0x403c9700,len:0xc08
load:0x403cc700,len:0x2a0c
entry 0x403c98dc
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4201479f  PS      : 0x00060f30  A0      : 0x82002e1d  A1      : 0x3fceb280  
A2      : 0x00000000  A3      : 0x00000087  A4      : 0x000000f0  A5      : 0x000000ef  
A6      : 0x00000000  A7      : 0x000000f0  A8      : 0x82020b50  A9      : 0x3fceb200  
A10     : 0x3fcf1c10  A11     : 0x00000000  A12     : 0x000000f0  A13     : 0x00000000  
A14     : 0x3fcf1bc4  A15     : 0x000000ff  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000003c  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffc  

Backtrace: 0x4201479c:0x3fceb280 0x42002e1a:0x3fceb2a0 0x42032f2a:0x3fceb2f0 0x40377933:0x3fceb320 0x403cd86b:0x3fceb350 0x403cdb2a:0x3fceb380 0x403c9952:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710```

bmorcelli, 2025-03-03:
I am probably missing something.. your log is very good.. I'll print the decoded version here with the next change when I get home

bmorcelli, 2025-03-03:
I turned on Logs for troubleshooting and changed a setting for the board.. if you could test with oone of the boars and paste the Serial logs again (and tell me which board are the logs)

Shadowtrance, 2025-03-03:
Yeah i have no idea what's going on at this point tbh.
It just doesn't seem to want to boot for whatever reason... consider me confused. 🤣
This is from the *043 board. no doubt the *550 would be the same as they're basically the same (same esp32s3 module onboard)

```
ELF file SHA256: b4f2fbd1cb4cf374

E (398) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377bd0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x518
load:0x403c9700,len:0xc08
load:0x403cc700,len:0x2a0c
entry 0x403c98dc
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4201648b  PS      : 0x00060f30  A0      : 0x82003361  A1      : 0x3fceb280  
A2      : 0x00000000  A3      : 0x00000087  A4      : 0x000000f0  A5      : 0x000000ef  
A6      : 0x00000000  A7      : 0x000000f0  A8      : 0x82025c80  A9      : 0x3fceb200  
A10     : 0x3fcf1c10  A11     : 0x00000000  A12     : 0x000000f0  A13     : 0x00000000  
A14     : 0x3fcf1bc4  A15     : 0x000000ff  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000003c  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd  

Backtrace: 0x42016488:0x3fceb280 0x4200335e:0x3fceb2a0 0x4203a342:0x3fceb2f0 0x40377933:0x3fceb320 0x403cd86b:0x3fceb350 0x403cdb2a:0x3fceb380 0x403c9952:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710```

bmorcelli, 2025-03-03:
yes, I am very confused too.. 
Now i've found some errors here, I wasn't starting the Flags properly, but it wasn't supposed to be showing these errors...

I am using [this board](https://github.com/rzeldent/platformio-espressif32-sunton/blob/ff4ef01c23cdb6ffe1e0033c2c62fd76a8a1b0c0/esp32-8048S043C.json) file, pointing to the same pins_arduino.h content file..

Launcher uses a customized version of Arduino-Esp32 2.17 version.. but it isn't supposed to be a problem, as it is working on all other boards (ESP32 and S3s)

well.. I uploaded a new version with some flags fixed and increased the coredump partition.

need a new test

bmorcelli, 2025-03-04:
I moved these 2 boards to:  BetaRelease > WIP.. didn't try anything diffent yet.. but I'm curious if the **BetaRelease > ESP32 > ESP32-S3-16mb** will work.. 

it is a tft-less version, where you can launch the firmwares from the webUi, the informations you get from the Serial Monitor

bmorcelli, 2025-03-04:
Turns out it was a problem initializing the touch driver.. 

I tried it on a generic ESP32-S3 to check if I get the same error, and it got traced right to the bb_captouch initialization.

Now I passed through this error and now I need to check if it works with the display.

Shadowtrance, 2025-03-04:
Just tried the latest wip on the *043 and still crashing after flashing (and unplugging/plugging back in).
And the generic ESP32-S3 16mb didn't work. Similar result with just crashing but no backtrace.
This is super freaking weird...

```
ELF file SHA256: bc6fbaf9c6eec3ff

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420ca566
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x518
load:0x403c9700,len:0xc08
load:0x403cc700,len:0x2a0c
entry 0x403c98dc
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4038079c  PS      : 0x00060c33  A0      : 0x80383744  A1      : 0x3fceba10  
A2      : 0x10101010  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060c23  
A6      : 0x00060c20  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0xffffffff  
A10     : 0x3fca0298  A11     : 0x3fcec40c  A12     : 0xfffffffe  A13     : 0xfffffffd  
A14     : 0xd3101010  A15     : 0x00ffffff  SAR     : 0x00000017  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x10101010  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  

Backtrace: 0x40380799:0x3fceba10 0x40383741:0x3fceba50 0x403838d1:0x3fceba70 0x40378613:0x3fceba90 0x403789bd:0x3fcebac0 0x420564e0:0x3fcebae0 0x4204b2f0:0x3fcebb30 0x4201ccf9:0x3fcebba0 0x4201ce02:0x3fcebc80 0x42005e61:0x3fcebca0 0x42022222:0x3fcebcf0
```

bmorcelli, 2025-03-04:
Hey man.. I Changed the touchdriver Ifrom bb_captouch to TouchLib, and it seems to be working on an other board with this driver..

can you test it please??

bmorcelli, 2025-03-05:
If you build locally, and use VSCode Serial Monitor, it will give the exact point it is breaking

Shadowtrance, 2025-03-05:
Yeah that's why i wanted to try building locally. i tried the other day but it just threw a bunch of errors at me. :(
Starting fresh this time (wiped out the .platformio folder) and hopefully that works. 🤞

Shadowtrance, 2025-03-05:
Success! And not success. 🤣 It builds!
Looks like it's failing at initializing the screen.
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x518
load:0x403c9700,len:0xc08
load:0x403cc700,len:0x2a0c
entry 0x403c98dc
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 1
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4038079c  PS      : 0x00060c33  A0      : 0x80383744  A1      : 0x3fceba10  
A2      : 0x10101010  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x00060c23  
A6      : 0x00060c20  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0xffffffff  
A10     : 0x3fca02a8  A11     : 0x3fcec40c  A12     : 0xfffffffe  A13     : 0xfffffffd  
A14     : 0xd3101010  A15     : 0x00ffffff  SAR     : 0x00000017  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x10101010  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  

Backtrace: 0x40380799:0x3fceba10 0x40383741:0x3fceba50 0x403838d1:0x3fceba70 0x40378613:0x3fceba90 0x403789bd:0x3fcebac0 0x42056004:0x3fcebae0 0x4204ae14:0x3fcebb30 0x4201d0d9:0x3fcebba0 0x4201d1e2:0x3fcebc80 0x420061b5:0x3fcebca0 0x42021d7a:0x3fcebcf0
  #0  0x40380799 in compare_and_set_native at /home/runner/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at /home/runner/esp32-arduino-lib-builder/esp-idf/components/esp_hw_support/include/soc/spinlock.h:103
      (inlined by) xPortEnterCriticalTimeout at /home/runner/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/port.c:301
  #1  0x40383741 in vPortEnterCritical at /home/runner/esp32-arduino-lib-builder/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:578 (discriminator 1)
      (inlined by) multi_heap_internal_lock at /home/runner/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:142 (discriminator 1)
  #2  0x403838d1 in multi_heap_malloc at /home/runner/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:233
      (inlined by) multi_heap_malloc at /home/runner/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:223
  #3  0x40378613 in heap_caps_malloc_base at /home/runner/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:175
      (inlined by) heap_caps_malloc_base at /home/runner/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:120
  #4  0x403789bd in heap_caps_calloc_base at /home/runner/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:487
      (inlined by) heap_caps_calloc at /home/runner/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:496
  #5  0x42056004 in gdma_new_channel at /home/runner/esp32-arduino-lib-builder/esp-idf/components/driver/gdma.c:142
  #6  0x4204ae14 in lcd_rgb_panel_create_trans_link at /home/runner/esp32-arduino-lib-builder/esp-idf/components/esp_lcd/src/esp_lcd_rgb_panel.c:491
      (inlined by) esp_lcd_new_rgb_panel at /home/runner/esp32-arduino-lib-builder/esp-idf/components/esp_lcd/src/esp_lcd_rgb_panel.c:178
  #7  0x4201d0d9 in Arduino_ESP32RGBPanel::getFrameBuffer(short, short) at .pio/libdeps/CYD-8048S043C/GFX Library for Arduino/src/databus/Arduino_ESP32RGBPanel.cpp:140
  #8  0x4201d1e2 in Arduino_RGB_Display::begin(int) at .pio/libdeps/CYD-8048S043C/GFX Library for Arduino/src/display/Arduino_RGB_Display.cpp:66
  #9  0x420061b5 in setup() at src/main.cpp:306 (discriminator 3)
  #10 0x42021d7a in loopTask(void*) at C:/Users/teron/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

Shadowtrance, 2025-03-05:
From what i can see it *looks* like the RGB_PANEL (and therefore the other related flags) aren't enabling. although that could just be vscode and platformio being weird too as it does sometimes...
[image]

bmorcelli, 2025-03-05:
I am preparing the flags externally, in a script that runs before thw compiling process

https://github.com/bmorcelli/M5Stick-Launcher/blob/90d47dfc1f15ec49e16857f10664c6ccc35be1e9/boards/CYD-2432S028/custom_flags.py#L117

But it should be showing fine, if you select the right env.

https://github.com/bmorcelli/M5Stick-Launcher/blob/90d47dfc1f15ec49e16857f10664c6ccc35be1e9/boards/CYD-2432S028/platformio.ini#L154

Shadowtrance, 2025-03-05:
Yeah i saw that. didn't know if it was working correctly or not as I'm used to things lighting up if the flag define works.
And yeah I've selected the correct env from the platformio menu which builds the selected env (043 board in this case).
Either way it's failing at initializing the screen as i logged in the previous comment. Why? i have no idea, i can't figure out why.

bmorcelli, 2025-03-05:
It is mimicking the same setup of ArduinoGFX setup for screen test..

https://github.com/moononournation/Arduino_GFX/blob/62d5e061ae3e2682a9001c88bcb41e20602b4fef/examples/PDQgraphicstest/Arduino_GFX_dev_device.h#L224

Have you tried this PDQgraphicstest, to see if it lights up?

buckminster-fuller, 2025-03-05:
> I am preparing to port to many of these Sunton boards (Cyd variants), so I'm changing a few things to better handle the tons of environment..
>
> It is mostly missing the touchscreen for the devices [...]

Excellent work mate! 

I´ve got a ESP32-3248S035C (3.5" ST7796  display with capacitive touch GT911) 

Let me know if I can help somehow!

Capacitative Touch: INT = 21, SCL = 32, SDA = 33, RST = 25 (GT911)

LED1 has three GPIO pins : RED = 4, GREEN = 16, BLUE = 17

BACKLIGHT = GPIO 27

Careful with Capacitive Touch - pin 21 has two other definitions (min)

SCREEN SPI: DC = 2, SCLK = 14, MOSI = 13, MISO = 12, CS = 15 RST = EN(-1)

bmorcelli, 2025-03-06:
@buckminster-fuller BetaRelease > WIP > CYD-3248S035C and CYD-3248S035R

can you test it? and open a new issue to discuss it there

bmorcelli, 2025-03-10:
JC3248W535C available on WIP for testing

buckminster-fuller, 2025-03-12:
> it is not on Bruce..
> 
> https://bmorcelli.github.io/M5Stick-Launcher/
> 
> Beta Release > WIP > CYD-3248W535C

Apologies, I’m juggling too many things at the moment and got confused. 
I´d give it a try but I actually only got an ESP32-3248S035C. 
Speaking of bruce, I saw you forked this repository and I´m wondering if you are porting bruce for various CYD variants, too.

bmorcelli, 2025-03-27:
New update on 8048W550C board.. now touch an tft are working.. can you test it on the 043C too??