bmorcelli/Launcher #110: SD Card não reconhece

7iagoCabral, 2025-02-07:
Olá, estou com um problema ao utilizar o Launcher na placa amarela.
Placa ESP32-2432S028 com saída microusb única e sem o CI de memoria `25Q32 4MB` (Localizado abaixo do ESP32)
SD card utilizado Western Digital class 10 32GB

- Já reduzi a partição do cartão para 4 GB na tentativa de reconhecer o cartão
- Já tentei instalar outro firmware e também não funciona o leitor
- O sd esta formatado em FAT32

Li que a placa so tem 2 portas SPI que é utilizada para Display, Touch e Slot SD e uma delas tem que ser virtualizada via software. E poderia ser este o problema.
Minha última suspeita é que a placa veio com defeito.

bmorcelli, 2025-02-07:
E aí bicho, blz?

Cara, se nem o Launcher, nem Marauder ou outro firmware reconhece o cartão SD, o problema pode ser o cartão mesmo.. 

Eu sempre recomendo SDHC, menor que 32gb, formatado em Fat32 (Se o teu é classe10, provavelmente é um SDXC).. eu nunca tentei particionar ele e tal.. mas eu acho que ele só reconheceria a partição FAT32 se ela estiver no inicio do particionamento..

A velocidade de escrita/leitura dos ESP32 também não é lá aquelas coisas, então não adianta investir em um cartão caro.. pega um baratinho, de 8gb que tá ótimo pra ele, tem desses por 15 reais pelo choices no Ali

7iagoCabral, 2025-02-07:
Vou comprar outro SD card e testar. Tomara que seja o cartão. caso contrario será o hardware ;(

7iagoCabral, 2025-02-11:
Estranhamente consegui usando mesmo cartão micro SD.
Tentei utilizar em outro lugar e o cartão corrompeu, então formatei pelo `Raspeberry Pi Imager` (coisa que tinha feito antes) simplesmente a placa amarela reconheceu.

closed 2025-02-11 by 7iagoCabral