bmorcelli/Launcher #111: Esp32 marauder v4 og

Amking123, 2025-02-08:
Hi I have a question why is my esp32 marauder v4 og cannot run the M5 luncher ?
It's TFT2.8 240×320 V1.2 I made it with ESP32 wroom32 type c but it only works with the marauder v4 og firmware and it only shows the main loading luncher and it becomes frozen when you turn it on [image]

bmorcelli, 2025-02-08:
Yeah.. this issue showed up when I changed the way I handled the Input handler.. somehow it broke the touchscreen..

Can you try the version 2.4.1, please? Download the .bin file from releases https://github.com/bmorcelli/M5Stick-Launcher/releases/tag/2.4.1

And install it with https://web.esphome.io

Amking123, 2025-02-08:
Thanks I will give it a try

Amking123, 2025-02-08:
Sorry but I have a question it's not about the version that you said but what is the differences between a CYD module and a marauder v4 og you know there are same 2.8 TFT and esp32 wroom32 but why when I tried the CYD firmwares I only get a white screen

bmorcelli, 2025-02-08:
They probably use different pins to connect to the display..

Steuriger, 2025-02-08:
Hey im also trying to do it like that rn but im having trouble flashing it when i go to https://web.esphome.io/ and upload the https://github.com/bmorcelli/M5Stick-Launcher/releases/download/2.4.1/Launcher-Marauder-v4-OG.bin flashed it but when I open Putty i get errors and it doesn't work...

Steuriger, 2025-02-09:
Got it to work now i can enter the launcher but then can't choose any option and the SD card is also not recognised i used those connections:
https://raw.githubusercontent.com/justcallmekoko/ESP32Marauder/refs/heads/master/schematics/Schematic_ESP32-Marauder-2_ESP32-Marauder-2-Schematic_20191007113616_png.png 
and for the SD card i used the configs found here:
https://github.com/bmorcelli/M5Stick-Launcher/blob/main/boards/marauder-v4og/platformio.ini

Amking123, 2025-02-09:
Thanks it worked perfectly but now I only can install the marauder v4 og and others like Bruce I can not install them

Amking123, 2025-02-09:
Sorry but is there any way to run the CYD firmware on the marauder v4 og

Steuriger, 2025-02-09:
> Thanks it worked perfectly but now I only can install the marauder v4 og and others like Bruce I can not install them

What Connections did you use does the SD Card Work aswell?

Amking123, 2025-02-09:
I found it in YouTube 
[image]

Steuriger, 2025-02-09:
It still doesn't work when I use Putty again i get those errors:
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 1
E (602) esp_core_dump_flash: N▒:▒ɕ dump partition found!
E (602) esp_core_dump_flash: No core dump partition found!
dutyCycle for bright 0-255: 255[   795][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[   801][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  1111][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
I flashed [this file](https://github.com/bmorcelli/M5Stick-Launcher/releases/download/2.4.1/Launcher-Marauder-v4-OG.bin) using the [ESP home flasher](https://web.esphome.io/)

Amking123, 2025-02-09:
What is the type of your esp 32

Steuriger, 2025-02-09:
> What is the type of your esp 32

ESP32 Wroom 32u

Amking123, 2025-02-09:
I have the same board. Try to connect the SD card into your TFT.

Amking123, 2025-02-09:
If you have any other micro SD card/SD card modules try to connect them and use 3.3v modules not 5v

Amking123, 2025-02-09:
I have a question do you know about the CYD modules?

Steuriger, 2025-02-09:
[image]
Maybe that helps and i have it connected to 3.3V

Amking123, 2025-02-09:
There is a connection for CYD in internet but I don't understand what dose _1 means
[image]

Steuriger, 2025-02-09:
Pin -1 means its Not connected i also Tried to make the cyd Connections but they didnt Work either for me but maybe the flashing was the Problem again

Amking123, 2025-02-09:
Why when I upload it the display show the loading screen for a second and then it disappear and a white screen appear and also the D2 and PWR lights on my module turn on at the same time you know when I upload the Marauder v4 og cod only the red light becomes on but when I upload a CYD code both red and blue light on my esp32 turn on

Steuriger, 2025-02-09:
Idk man but does the SD Card Work for you?

Amking123, 2025-02-09:
Yeah it works fine I even install a marauder on it

Steuriger, 2025-02-09:
What flasher did you use and how did you proceed?

Amking123, 2025-02-09:
I used the things that Mr Pirata said the exact thing and the exact connections and can you say how much is the storage of your SD card try 8 GB

Steuriger, 2025-02-09:
My sd card is 32GB but i always get the errors mentioned before when finished flashing did you preset those partitions or just go to https://web.esphome.io/ then connect and choose the file and flash?

Amking123, 2025-02-09:
I watched a video Wich said that before flashing hold the boot button and then connect the USB to your laptop

Steuriger, 2025-02-09:
Yeah thats normal you Always have to do that But did you follow a certain Tutorial and If so could you please Share the Link

Steuriger, 2025-02-09:
Oke i can try that but Tried IT with the esp32 wroom 32u so i think the Same as yours...

Amking123, 2025-02-09:
I have question does your esp32 has an external antenna

Steuriger, 2025-02-09:
Yes My esp32 Supports a external antenna via cable

Amking123, 2025-02-09:
[image]

Steuriger, 2025-02-09:
I got that exact one too i can try that one

Amking123, 2025-02-09:
With the M5 launcher for marauder v4 og it's pretty good but you can not install other firmwares you can only install the marauder

Steuriger, 2025-02-09:
Oke i now tried it with the exact same esp32 as you and yet still doesn't work still getting the same errors

Amking123, 2025-02-09:
I think you but if you don't know go to your device manager and see that you're board is recognized or not

Amking123, 2025-02-09:
And could you please send me a picture of that error

Steuriger, 2025-02-09:
[image]

[image]

Amking123, 2025-02-09:
Sorry I'm asking you but are you going to install marauder on it?

Amking123, 2025-02-09:
I mean on your launcher

Amking123, 2025-02-09:
Is it only appears on that website?

Steuriger, 2025-02-09:
Im Going to Install marauder Bruce and maybe Ghost ESP and what do you mean by IT only appears in the Website?

Amking123, 2025-02-09:
You can not install them I tried and it didn't work I mean there are so many ways to flash them like web or the flasher.exe and esp_idf

Steuriger, 2025-02-09:
You Just have to Compile the Files yourself

Amking123, 2025-02-09:
I didn't get but I downloaded the bin files and when I press the install button it said installed successful but nothing happened and Bruce and esp ghost are for CYD

Steuriger, 2025-02-09:
Yeah because of the Connections thats why you need build the bin yourself so you can Change the pins

Amking123, 2025-02-09:
Oh now I get it but I can not change a bin file

Amking123, 2025-02-09:
Hey do you have a telegram account?

Steuriger, 2025-02-09:
Yes i have one you want to Chat there?

Amking123, 2025-02-09:
Yes if you can

Amking123, 2025-02-09:
This is my account massage me 
@Amking_123

Steuriger, 2025-02-10:
@bmorcelli Is there any way you know what i do wrong when flashing?

bmorcelli, 2025-02-10:
After you flashed, the screen worked well, right?

In 2.4.1 version, navigation is only through the bottom screen buttons (Prev, Sel, next).. if it doesn't work, try pressing the same spots but at the top of the screen (maybe this screen rotation/coordinates are different)

What is the problem you are getting?

Steuriger, 2025-02-10:
Oke small update it works now the buttons are just on top of the screen and i can only flash from the SD card if i after starting with it pluged in i unplug it and then plug it in again else it says no SD card found but thank you so much for your help!

Steuriger, 2025-02-10:
Oke i tried it with this esp32 and it worked.

[image]

Now i tried it with an esp32 wroom 32u and it doesn't work.

bmorcelli, 2025-02-10:
Make sure GPIO numbers are the same.. it is important to see Serial Monitor logs to understand what is going on

Steuriger, 2025-02-10:
The GPIO pins should be the same if i get some answers i will comment here again.

Amking123, 2025-02-10:
Sorry but Mr Pirata if I change the pins connections to the CYD display is there any changes if running the code?

bmorcelli, 2025-02-10:
Maybe, idk.. the display driver is supposed to be the same..

Steuriger, 2025-02-11:
Did not change a single thing but works (on the esp32 wroom 32u) now so IDK

bmorcelli, 2025-02-16:
Hey man.. I tried to fix this devide today, can you test the BetaRelease for me, please?

[WebFlasher ](https://bmorcelli.github.io/M5Stick-Launcher/) > Marauder > V4-OG

closed 2025-02-19 by bmorcelli