bmorcelli/Launcher #112: Add support of the CYB 2.2 inch capacitive screen. ESP32-2432S022

T-REX-XP, 2025-02-12:
Could you please add support for the CYB ESP32-2432S022 board?

**Documentation and examples:** [link](http://pan.jczn1688.com/1/ESP32%20module)

```
Serial port WebSerial VendorID 0x1a86 ProductID 0x7523
Connecting...
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision 3)
Features: Wi-Fi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: ac:15:18:d4:59:d4
```
[image]
[image]

Thanks in advance

bmorcelli, 2025-02-16:
Hey man... I've foud this project that might work in the TFT configuration 

https://github.com/HASwitchPlate/openHASP/pull/838

It is very hard for me to bring it to Launcher since I don't have the device.. 

Pleas go to the [Launcher Flasher](https://bmorcelli.github.io/M5Stick-Launcher/) and try the BetaRelease > CYD-2432S022C version and tell me the results, please

msahin, 2025-02-17:
[image]
i have uploaded  beta build for this board but screen not working. only white screen. and these messages showing up on terminal

bmorcelli, 2025-02-17:
https://github.com/Bodmer/TFT_eSPI/issues/3281

I beleive you'll have a lot of trouble with this display..

msahin, 2025-02-18:
the only thing working for this board is OpenHASP with these (https://github.com/HASwitchPlate/openHASP/blob/master/user_setups/esp32/esp32-2432s022c.ini) settings. i think everything works. but I would like to use marauder or m5 launcher with this board.

bmorcelli, 2025-02-18:
Yes.. I saw that.. seems like TFT_eSPI (Library used in this project, Marauder and Bruce) can't make it work..

In openHasp it is using LovyanGFX library and LVGL..

bmorcelli, 2025-02-24:
hey man.. I'm trying a new graphics lib (Arduino_GFX), can you test it, and tell me if it works?

Please go to [Launcher Flasher](https://bmorcelli.github.io/M5Stick-Launcher/) > BetaRelease > CYD-2432S022C (Capacitive)

msahin, 2025-02-24:
[image]

i flashed and got only white screen and these serial output. bright white screen when it turns on. then the backlight dims. when i touch the screen the light increases. i think it's energy saving and touch works but no content on display

T-REX-XP, 2025-02-24:
> Pleas go to the [Launcher Flasher](https://bmorcelli.github.io/M5Stick-Launcher/) and try the BetaRelease > CYD-2432S022C version and tell me the results, please

Hi. I flashed the device, but the result was the same as in the comment above.

bmorcelli, 2025-02-24:
Yes, ArduinoGFX and TFT_eSPI don't handle gpio over 31 for the tft pins.. the only library which works is LovyanGFX..

I'm thinking of a way to integrate it with the code I already have and keep it easy to add new boards..

I hope I'll have something until the end of this week..

bmorcelli, 2025-02-24:
I updated now, can you test it?

now  CYD-2432S022C is using LovyanGFX lib.. using the same openHASP settings

[Launcher Flasher](https://bmorcelli.github.io/M5Stick-Launcher/) > BetaRelease > CYD-2432S022C (Capacitive)

msahin, 2025-02-24:
ELF file SHA256: 45bf3835fc852bd1

E (241) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1276
ho 0 tail 12 room 4
load:0x40078000,len:13232
load:0x40080400,len:3076
entry 0x400805f0
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
E (520) esp_core_dump_flash: ��ɕ�dump partition found!
E (520) esp_core_dump_flash: No core dump partition found!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d6c9a  PS      : 0x00060830  A0      : 0x800f0de5  A1      : 0x3ffb2240  
A2      : 0x3ffc432c  A3      : 0x3ffc4466  A4      : 0x3ffc4418  A5      : 0x3f415f50  
A6      : 0x00000032  A7      : 0x3ffbdb70  A8      : 0x800d6c91  A9      : 0x3ffb2220  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000001  A13     : 0x3ffb2200  
A14     : 0x3ffc8490  A15     : 0x3ffb2210  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008bfdd  LEND    : 0x4008bfed  LCOUNT  : 0xffffffff  

Backtrace: 0x400d6c97:0x3ffb2240 0x400f0de2:0x3ffb2290

looping this. no screen

[esp-web-tools-logs.txt](https://github.com/user-attachments/files/18947797/esp-web-tools-logs.txt)

bmorcelli, 2025-02-24:
Found the problem and fixed it.. I wasn't starting the object in the right way for this library, now it is fine

I've tested it on my T-Display S3, which uses 8bit_parallel too with lovyan and now it is supposed to be initializinf and working.

[Launcher Flasher](https://bmorcelli.github.io/M5Stick-Launcher/) > BetaRelease > CYD-2432S022C (Capacitive)

bmorcelli, 2025-02-25:
It overflowed the partition size.. can you try it again, please?

msahin, 2025-02-25:
thank you. it is working fine but i think the screen orientation is wrong

https://github.com/user-attachments/assets/6acc5929-4985-43c7-b83d-d02c7f583fc7

bmorcelli, 2025-02-25:
Niiiice!!! Awesome!! good to see the display working.. It might be a problem of the cfg.offset_rotation that was set to 3 (same used in TFT_eSPI), but maybe it should start at 0 (as an offset), and default ROTATION kept in 3 for the touch coordinates work well..

https://github.com/bmorcelli/M5Stick-Launcher/commit/5c031e46b78f5d07a87ff88f45c8b84455a1a95e

I updated now, you can test it

msahin, 2025-02-25:
[image]
[image]
[image]
[image]

everything works but i cant get connected to the wifi because keyboard characters are weird. wifi ap works. I tried to enter wifi credentials from wifi ap but I didn't succeed there either.

bmorcelli, 2025-02-25:
rebuilding.. i broke a few things..
available in 6 minutes

msahin, 2025-02-26:
As far as I have tried, everything works perfectly. Thank you for your efforts.

I'm just wondering, there is a button on the board but I guess it has no function, can this button be assigned a function?

bmorcelli, 2025-02-26:
> As far as I have tried, everything works perfectly. Thank you for your efforts.

Thank you!

> I'm just wondering, there is a button on the board but I guess it has no function, can this button be assigned a function?

It could, but it's not necessary gor launcher 

> and i hope this card becomes compatible with other projects for example Marauder, bruce

Maybe..

msahin, 2025-02-26:
I was able to successfully upload and install openhasp .bin file. now dual booting launcher and openHASP. it's so cool. Thank you 

m5launcher works very well and the card does not heat up, but when openHASP is running, the card gets very hot. I think your code is more optimized.

bmorcelli, 2025-02-26:
It happens because OpenHasp keeps WiFi turned On the whole time.. it heats the ESP32 chip too much..

Launcher only uses WiFi while in OTA (for M5 devices) and while WebUI is on

closed 2025-02-27 by bmorcelli

mike-logic, 2025-03-15:
> I'm just wondering, there is a button on the board but I guess it has no function, can this button be assigned a function?

I wondered about this too but i think it's just the interrupt for the battery connection right next to it.

daerevets, 2025-03-19:
I hope I have not misunderstood but the button turns the device on and off when using the battery. The trick however, is that you have to press and hold it at least 10 seconds or longer for it to turn off. 
Maybe real programmers can help me but I think this button code is built into the ESP32. I say this because I tried re-flashing and broke it as it now boots to a blank white screen. 
 However, the on-off button still works.
One more thing, you can press and hold this button while on usb (with or without a battery) and after the same long wait it simply goes off and then immediately boots on again. Presently in my case to my now blank white screen. ;(

daerevets, 2025-03-19:
Not sure why my comment above comes up as "daerevets"? 

Can someone look into this. I logged in as "sd_read@hotmail.com".

sd-read, 2025-03-19:
OK, I am the same sd_read@hotmail.com above but have signed in with an old account that I forgot I even had. 
And this account is "sd_read@yahoo.ca" and it seems to be working correctly.

sd-read, 2025-03-19:
I have a question about the Launcher. I am trying to log into my network but I guess my cap touch screen is not calibrated so I cannot get the right passcode in. It is all digits and I need to press above the numbers along the top (press where the numbers show on the screen) but when I make a mistake I can not press "DEL" as it is 'virtually' above the screen. 
Is there something I can do to get my password in? If I could get through the 12 digits just once .....

bmorcelli, 2025-03-19:
You can Edit the file config.conf in your SDCard and add your SSID and password there, so your device will try to connect to it automatically when you choose a known ssid.

bmorcelli, 2025-03-19:
Touch calibration is something I need to work on.. it is something that really stress me out 😊

mike-logic, 2025-03-19:
...out of curiosity...but what firmware are we running on our 2432S022s? i can't really seem to find any that are meant for it...other than maybe WifiSpider

sd-read, 2025-03-19:
Thank you bmorcelli, that worked! 

And thank you for producing Launcher, it blew my mind.