bmorcelli/Launcher #113: SD CARD

GuMAZ777, 2025-02-14:
WHEN USING LAUNCHER ON T EMBED CC1101 THE SD CARD KEEPS NOT BEING READABLE, AFTER BEING READABLE. AND SOMETIMES THE OTA ASWELL. I REALLY LIKE USING IT FOR THE BRUCE AND FACTORY LILYGO  FW, SO I CAN SHUT THE T EMBED DOWN WHEN NOT IN USE, USING THE LILYGO FW. SINCE THE BRUCE IS ONLY ABLE TO PUT IT TO SLEEP. IS THERE ANY SOLUTION FOR THE SD CARD. IT HAS BEEN FORMATTED. FAT32. THANK YOU.

bmorcelli, 2025-02-17:
I really don't understand why some SDCards have this behavior, because all my sdcards work fine and I don't have this issue, wo it is really hard to debug an error that I can't replicate.

The Full turn off is a problem that I maybe I can add, although I think it is useless because you'll be slave of a charger to turn on the device, as the power button is hidden below the battery, inaccessible from the outside..

GuMAZ777, 2025-02-17:
It would be nice to have the complete power off as a option, for when not in use. We're more a slave to the charger without it. After all there is sleep. Which should be what shut down is now. To implement full shut down for shut down would defile it's true function. Just let us know when you hit shut down well have to use the charger to restart it. It also helps against the battery degrading aswell i think. I switched my sd card to a 16g. I was using a 32g. Not sure if that was the issue. I'm monitering it thou. Thank you for your reply. And I'm enjoying your software

bmorcelli, 2025-02-17:
Cc1101 and PN532 are draining battery while in deepsleep mode.. I will add this full turn off command on CFG, let's see how it works

bmorcelli, 2025-02-19:
This problem was fixed here https://github.com/bmorcelli/M5Stick-Launcher/pull/116 by @TheDuc (tyvm!)

Test it on BetaRelease before I release a new version, please

closed 2025-02-20 by bmorcelli