bmorcelli/Launcher #114: liligo t dongle

Erwantest123, 2025-02-18:
hello bmorcelli is it possible to make a version for the liligo t dongle card and its button included.
==
Long press to select and short press to choose.

bye @Erwantest123

bmorcelli, 2025-02-18:
You can run it without the display for T-Eongle S3.. 

install the ESP32>ESP32-s3 16Mb, after install, remove it from USB and plug again and check the Serial Monitor..

It will start the WebUI in Access point mode automatically at the first time, so there you can configure your network, if you want, and install/update firmware wirelessly..

After install the first firmware woth Launcher, to get in the Launcher again, you meed to plung into the USB, and press its button within 5s (dont plug with button pressed, otherwise it'll start in download mode)... So then you can repeat the process

Erwantest123, 2025-02-18:
ok

closed 2025-02-20 by Erwantest123