bmorcelli/Launcher #115: circuitmess chatter 2.0 support request

Poon-Harpoon, 2025-02-19:
circuitmess chatter 2.0 support request it supports chatter, meshtastic but theres so much more this esp32 is capable of can we get some live since these come with lora screen phone based keyboard for only about $10ea

bmorcelli, 2025-02-20:
Interesting device.. I couldn't find a project using TFT_eSPI library that I couldn't easily copy the display setup and input setup, so if you have this information to share, it would be a great help!!

bmorcelli, 2025-02-20:
https://community.circuitmess.com/t/chatter-screen-pins-for-tft-espi-library/5349

One step further..

Poon-Harpoon, 2025-02-21:
Update I just checked I thought it has an sd micro card slot it doesnt but im pretty sure it wouldnt be hard to add one schematic 

[image]

Poon-Harpoon, 2025-02-21:
https://github.com/CircuitMess/Chatter2-Firmware/