bmorcelli/Launcher #120: Error gpio_set_level(227) after setting the SDCard pins in ESP32-S

NainrousRC-X, 2025-03-03:
I have an ESP32-S "NodeMCU-32 V1.3"

I am using the online installer selecting "ESP32-4mb" V2.4.4

Everything seems to work fine, until I establish the SDCard pins, after restarting the ESP32, checking from the serial terminal it starts saying the following:

E (88) gpio: gpio_set_level(227): GPIO output gpio_num error

then the "Launcher" banner appears and that message appears again on several lines, then I get the IP to connect through the generated WiFi and when I log in after loading the web interface it freezes and if I reload the page it tells me "This site cannot be accessed"

Checking the serial terminal I noticed that the same message appeared again but this time so many times that it seemed that an auto DDOS was performed, the final message is:

E (721025) gpio: gpio_set_level(227): GPIO output gpio_num error

I clarify that I am establishing the connection of the Micro SDCard adapter as follows:

MOSI → GPIO 23
MISO → GPIO 19
SCK → GPIO 18
CS → GPIO 12
GND → GND
VCC → 3.3V

and YES, previously CS I put it on PIN 5 and it doesn't recognize it, it only recognizes it if I put it on PIN 12

The Micro SDCard is... "128MB" I have used this same one with the same PIN configurations in Marauder and I have not had any problems

bmorcelli, 2025-03-03:
my bad, I Forgot to change SDCAR_CS for the _cs variable in the code..

https://github.com/bmorcelli/M5Stick-Launcher/blob/19d128f0bad308a9df0486cd9156775196ef408f/src/sd_functions.cpp#L68

I will update it asap

bmorcelli, 2025-03-05:
Fixed in 2.4.5

closed 2025-03-05 by bmorcelli