bmorcelli/Launcher #124: M5cardputer only 95 firmware

erwantestI23, 2025-03-06:
[image]

[image]

[image]

It is normal

bmorcelli, 2025-03-06:
Yes, for now it is listening firmware with more than 200 downloads on M5Burner, because there is no paging system yet, and the lists are being too big for the Ram memory 

It will be restored in a few weeks

bmorcelli, 2025-03-06:
Something like that,

 I need to redo some backend stuff in order to make these things work as intended, as Launcher should not rely on external memory due to the partitioning (sometimes spiffs/LittleFS is not available)..

erwantestI23, 2025-03-07:
after it is not possible to create a new partition but small one of 500 kb ?

Partitions.csv
my_partition, data, custom, size, size

And 

File file = SPIFFS.open("/my_partition/file.json", FILE_WRITE);

erwantestI23, 2025-03-08:
I know! For the JSON, analyze half of it and display the first 50 firmware, then after 50 firmware, analyze the rest.

bmorcelli, 2025-03-08:
As I don't use a real database to make proper requests, I need to  break the final json file into multiple files and make the logic to:
* Resquest the number of db files (lists of 50, for example)
* Request the first file and allow navigation from 1 to 50
* If pass to 51, clear the json object and request the second file
* If come backwards, from 1 to last, or from 51 to 50, clear the object and download the new one

I can't rely on a file saved anywhere on Launcher, because some devices doesn't have SD card, or Spiffs partition is too small, or inexistant, and some devices doesn't have PSRam.. so things need to be done using RAM for it.

Im working on a solution to make the requests to a db server instead of raw json files on GitHub, so it will take some time to be released, because I need to develop the whole solution (I'm working on it)

closed 2025-03-28 by erwantestI23