bmorcelli/Launcher #125: (WIP) CYD-3248S035C (3.5in, Capacitive) - no display, reboots when touching certain area on display

buckminster-fuller, 2025-03-06:
### serial output:

`ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1276
ho 0 tail 12 room 4
load:0x40078000,len:13232
load:0x40080400,len:3076
entry 0x400805f0
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
E (522) esp_core_dump_flash: �� core dump partition found!
E (522) esp_core_dump_flash: No core dump partition found!
E (201) ledc: ledc_set_duty(725): LEDC is not initialized
E (202) ledc: ledc_update_duty(653): LEDC is not initialized
Touch IC Started
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1276
ho 0 tail 12 room 4
load:0x40078000,len:13232
load:0x40080400,len:3076
entry 0x400805f0
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
E (522) esp_core_dump_flash: �� core dump partition found!
E (522) esp_core_dump_flash: No core dump partition found!
E (201) ledc: ledc_set_duty(725): LEDC is not initialized
E (202) ledc: ledc_update_duty(653): LEDC is not initialized
Touch IC Started
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!`

bmorcelli, 2025-03-06:
Display starts?

buckminster-fuller, 2025-03-06:
No, display remains dark.

bmorcelli, 2025-03-07:
Ok, display settings were missing, so now it should be working (fingers crossed)

buckminster-fuller, 2025-03-07:
Indeed. Display starts. Launcher starts. Well done! 
I´ll play around with it later but for now I saw 2 issues:

- touch is working but needs some debounce (every single touch is being processed as double touch)
- WUI starts but after authentication the browser shows gibberish and launcher even reboots sometimes:
 

`Configuring Webserver ...
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e3cc3  PS      : 0x00060430  A0      : 0x800e3d4b  A1      : 0x3ffb3c20  
A2      : 0x00000000  A3      : 0x3ffd0820  A4      : 0x00000000  A5      : 0x3ffcc5ac  
A6      : 0x3ffcc5c4  A7      : 0x00060023  A8      : 0x8011c950  A9      : 0x3ffb3c10  
A10     : 0x00000000  A11     : 0x00000006  A12     : 0x00000014  A13     : 0x0000ffff  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x4008a600  LEND    : 0x4008a616  LCOUNT  : 0xffffffff  

Backtrace: 0x400e3cc0:0x3ffb3c20 0x400e3d48:0x3ffb3c50 0x4010f2d5:0x3ffb3c70 0x40115bce:0x3ffb3ce0 0x4011996e:0x3ffb3d10 0x4010a349:0x3ffb3d30

ELF file SHA256: c49d910de5a12c0a

E (1020) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1276
ho 0 tail 12 room 4
load:0x40078000,len:13232
load:0x40080400,len:3076
entry 0x400805f0
[boot] Turned on because (1= POWERON_RESET) (Other= Probably forced by launcher)--> 12
E (519) esp_core_dump_flash: N���ɕ�dump partition found!
E (519) esp_core_dump_flash: No core dump partition found!
E (478) ledc: ledc_set_duty(725): LEDC is not initialized
E (479) ledc: ledc_update_duty(653): LEDC is not initialized
Touch IC Started
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Press the button to enter the Launcher!
Configuring Webserver ...
> `

bmorcelli, 2025-03-07:
> touch is working but needs some debounce (every single touch is being processed as double touch)

I changed the debounce logic, maybe it is working better now

> WUI starts but after authentication the browser shows gibberish and launcher even reboots sometimes:

I believe it must be working fine now

closed 2025-03-07 by bmorcelli

buckminster-fuller, 2025-03-09:
ACK!